apache/nuttx #7078: LLVM Nuttx Toolchain

AndrewD, 2022-09-13:
We feel it would be useful to have Nuttx as a target for LLVM. Has anyone attempted this? It looks quite feasible to implement and it appears llvm has been successfully used for some target architectures (z80)

xiaoxiang781216, 2022-09-13:
> We feel it would be useful to have Nuttx as a target for LLVM. Has anyone attempted this? 

SIM on macOS use clang by default long time aga.
ARM/RISCV add clang support recently(https://github.com/apache/incubator-nuttx/pull/6105).

> It looks quite feasible to implement

Since many arch support clang, the work isn't huge for other arch I think. 

> and it appears llvm has been successfully used for some target architectures (z80)

Yes, it's important to use the modern compiler for old arch(e.g. z8, z80, ez80, z180 and z16), so we can relax c89 requirement: https://github.com/apache/incubator-nuttx/issues/6896.

acassis, 2022-09-14:
@AndrewD this is a great idea to use LLVM for Z80 with NuttX. I tried to use SDCC but it is failing: https://acassis.wordpress.com/2022/01/14/testing-nuttx-rtos-on-z80-simulator/ If you decided to include the clang for Z80 please let me know, I want to help!

acassis, 2022-09-14:
The LLVM Z80 port is here: https://github.com/jacobly0/llvm-project I hope he submit it to mainline to let more people start working on it!

AndrewD, 2022-09-15:
@acassis it's an fantastic exercise that people have added Z80 to LLVM as a target architecture. This is way more than I are proposing :) 

To be clear: our interest is in a LLVM cross toolchain for RISCV32 and potentially i686. The cleanest way to implement this appears to be to add Nuttx as a supported target OS.

Note that once Nuttx is available as a target OS for LLVM then any LLVM supported processor architecture could probably be used. With LLVM the same toolchain should be able to build for many Nuttx target processor architectures.

Our goal is to use LLVM on Windows to build applications that can be added to a live Nuttx system on our product. One way of explaining this is package management for Nuttx, where some Applications are build  separately from the Nuttx image that is deployed to the device.
The Nuttx image running on the device would be build on Linux and deployed "OTA" to the device.

The search for ananas in LLVM below shows how minimal a LLVM operating system port can be:
https://github.com/llvm/llvm-project/search?q=ananas
The Continuous Integration (CI) support for testing a LLVM operating system port also seems quite clean.

I just found this blog post explaining some of the details:
https://lf-net.org/blog/posts/2020-05-17-clang-osdev/

xiaoxiang781216, 2022-11-05:
Here is the work for arm clang

closed 2022-11-23 by pkarashchenko