apache/nuttx #7082: Change the flow code of function psock_setsockopt

vuhuyhop197, 2022-09-13:
We are implementing a USRSOCK for our WIFI chip. The issue is the function psock_setsockopt doesn't call usrsock_setsockopt. Because it prioritizes checking the protocol level. I think the code flow should be:
```
int psock_setsockopt(FAR struct socket *psock, int level, int option,
                     FAR const void *value, socklen_t value_len)
{
  int ret;
  /* Verify that the sockfd corresponds to valid, allocated socket */
  if (psock == NULL || psock->s_conn == NULL)
    {
      return -EBADF;
    }
#ifdef CONFIG_NET_USRSOCK
  if (psock->s_type == SOCK_USRSOCK_TYPE)
    {
      /* Handle setting of the socket option with usrsock */
      ret = usrsock_setsockopt(psock->s_conn, level,
                               option, value, value_len);
    }
  else
#endif /* CONFIG_NET_USRSOCK */
    {
      /* Handle setting of the socket option according to the level at which
      * option should be applied.
      */
      switch (level)
        {
          case SOL_SOCKET: /* Socket-level options (see include/sys/socket.h) */
            ret = psock_socketlevel_option(psock, option, value, value_len);
            break;
#ifdef CONFIG_NET_TCPPROTO_OPTIONS
          case IPPROTO_TCP:/* TCP protocol socket options (see include/netinet/tcp.h) */
            ret = tcp_setsockopt(psock, option, value, value_len);
            break;
#endif
#ifdef CONFIG_NET_UDPPROTO_OPTIONS
          case IPPROTO_UDP:/* UDP protocol socket options (see include/netinet/udp.h) */
            ret = udp_setsockopt(psock, option, value, value_len);
            break;
#endif
#ifdef CONFIG_NET_IPv4
          case IPPROTO_IP:/* TCP protocol socket options (see include/netinet/in.h) */
            ret = ipv4_setsockopt(psock, option, value, value_len);
            break;
#endif
#ifdef CONFIG_NET_IPv6
          case IPPROTO_IPV6:/* TCP protocol socket options (see include/netinet/in.h) */
            ret = ipv6_setsockopt(psock, option, value, value_len);
            break;
#endif
#ifdef CONFIG_NET_CANPROTO_OPTIONS
          case SOL_CAN_RAW:   /* CAN protocol socket options (see include/netpacket/can.h) */
            ret = can_setsockopt(psock, option, value, value_len);
            break;
#endif
          default:         /* The provided level is invalid */
            ret = -ENOPROTOOPT;
            break;
        }
    }
  return ret;
}
```

vuhuyhop197, 2022-09-13:
The issue is the same with psock_getsockopt

xiaoxiang781216, 2022-09-13:
which option do you want the wifi chip process first?

ethanlcz, 2022-09-13:
> think the code flow should be:

Could you please help to describe the issue you met in detail?
I think your patch might introduce other issues, for example, in origin design, psock->conn will log some common options (s_rcvtimeo, s_sndtimeo, etc.) and usrsock will check these settings before send request to usrsock server. Theses settings won't be logged in psock->conn with your patch and usrsock's behavior will be changed.

vuhuyhop197, 2022-09-13:
@xiaoxiang781216 In the case of enabling CONFIG_NET_USRSOCK, we want psock_setsockopt process usrsock_setsockopt first. Because we have written a wrapper for usrsock_setsockopt_handler.

vuhuyhop197, 2022-09-13:
The origin code flow: setsockopt -> psock_setsockopt -> psock_socketlevel_option. We want to change: setsockopt -> psock_setsockopt -> usrsock_setsockopt if CONFIG_NET_USRSOCK is enabled

acassis, 2022-09-15:
Hi @vuhuyhop197 I think you will need more checks here: imagine the scenario were the user have a WiFi with USRSOCK and an Ethernet in the board, this modification will always assume the user want to change the WiFi socket instead of the real interface underneath.

acassis, 2022-09-15:
@vuhuyhop197 I'm curious, what is your WIFI chip? Are you using ESP32? If so, are you planing to submit support to mainline?

vuhuyhop197, 2022-09-16:
@acassis Sorry, I can't share the info now.

closed 2022-09-17 by vuhuyhop197