apache/nuttx #7086: Any chance of getting netcat working on Nuttx 8.2?

greg5678, 2022-09-13:
So, it seems the Particle Photon's network functionality isn't working on the current version of Nuttx, but it does work on Nuttx 8.2. Is it possible to get netcat working on this older version of Nuttx on the photon? Just to see if it would work I took a copy of the current netcat folder from the nuttx-apps repo and put it in my `/apps/netutils` path. Then I selected it with `make menuconfig` and after running `make -j3 `surprisingly it seems to compile successfully. The thing is after taking the `nuttx.bin` binary and flashing it to the Particle Photon it booted up and I was able to get a shell, but didn't see netcat. So, since that didn't work is there a way to get netcat running on Nuttx 8.2?

acassis, 2022-09-13:
Hi @greg5678 unfortunately you cannot just copy a new directory from apps to old NuttX version neither vice-versa.
You need to adapt the Makefile/Make.defs/Kconfig and even the source file C (in the past was name_main() now is just main() name), after that you will get:
```
$ telnet 192.168.0.16
Trying 192.168.0.16...
Connected to 192.168.0.16.
Escape character is '^]'.

NuttShell (NSH)
nsh> ?       
help usage:  help [-v] [<cmd>]

  [           cmp         free        mb          pwd         time        
  ?           dirname     help        mkdir       rm          true        
  arp         dd          hexdump     mh          rmdir       uname       
  basename    df          ifconfig    mount       set         umount      
  break       echo        ifdown      mv          sh          unset       
  cat         exec        ifup        mw          sleep       usleep      
  cd          exit        kill        nslookup    test        xd          
  cp          false       ls          ps          telnetd     

Builtin Apps:
  netcat
  wapi
nsh> 
```
Kconfig: https://pastebin.com/raw/pU7urzR1
Make.defs: https://pastebin.com/raw/LTkeEmLq
Makefile: https://pastebin.com/raw/JPNGPrTh
Edite the netcat_main.c and rename the main function to netcat_main

Good luck!

acassis, 2022-09-13:
Also, don't forget to run: "make apps_distclean" to remove previous configuration, after that run "make menuconfig" to confirm everything is configured!

acassis, 2023-03-19:
User didn't reply! They don't care about our feelings! :-(

BTW, also duplicated here: https://github.com/apache/nuttx/issues/8839

Closing it!

closed 2023-03-19 by acassis