apache/nuttx #7105: libxx: Update libcxx to a newer version

AndrewD, 2022-09-16:
It would be useful to update Nuttx libcxx to [14.0.6](https://github.com/llvm/llvm-project/releases/tag/llvmorg-14.0.6) or even the latest  15.0.0 release to make newer C++ library features and fixes available. Even 15.0.0 (released Sep 6th) is worth considering.

At nuttx-10.4.0-RC0-213-g1b9c013dad NuttX applies three patches to libc++
12.0.0:
* 0001-Remove-the-locale-fallback-for-NuttX.patch
  - This is upstreamed at libc++ version 14.0.6
* 0001-libc-avoid-the-waring-__EXCEPTIONS-is-not-defined-ev.patch
  - Upstreamed at libc++ version 14.0.6
* 0001-libcxx-Rename-PS-macro-to-avoid-clashing-with-Xtensa.patch
  - Workaround for build failure on Xstensa, NFC on other targets, therefore
    not a significant issue
* 0002-Omit-atomic_-un-signed_lock_free-if-unsupported.patch
  - Upstreamed at libc++ version 14.0.6

This means that building LLVM 14.0.6's included libc++ should work (almost) without
patching.

xiaoxiang781216, 2022-09-16:
Yes, it's always good to update libcxx/libcxxabit to the last official release and remove the temp patch, could you send a PR?

AndrewD, 2022-10-12:
> Yes, it's always good to update libcxx/libcxxabit to the last official release and remove the temp patch, could you send a PR?

While we had a quick look in concept we haven't attempted to do the change and aren't sure if there are any other details we would need to resolve to do this. Do you have any documentation or hints as to how this was done previously?

xiaoxiang781216, 2022-10-12:
1. Change LIBCXX_VERSION(https://github.com/apache/incubator-nuttx/blob/master/libs/libxx/libcxx.defs#L21) and LIBCXXABI_VERSION(https://github.com/apache/incubator-nuttx/blob/master/libs/libxx/libcxxabi.defs)
2. Remove patch command from libcxx.defs and libcxxabi.def
3. Build if you are lucky, all is done. If compiler error happen, you have to look into and fix the error
4. Test sim:libcxxtest and your config to ensure the new library work as expect.

closed 2023-09-24 by pkarashchenko