apache/nuttx #7145: PPP: Use the DNS servers negotiated in the `IPCP` configuration

vshymanskyy, 2022-09-20:
DNS with PPP connections only work if `CONFIG_NETDB_DNSSERVER_IPv4ADDR` is manually set to DNS ip address (i.e. `0x08080808` for Google DNS).
IPCP could be used to configure network-provided DNS address.

acassis, 2022-09-22:
Good point, currently the IP is got automatically, so we need to add support to get the DNS as well.

duduita, 2023-07-25:
It seems that NuttX supports the usage of IPCP-provided DNS servers [by enabling the IPCP_GET_PRI_DNS and IPVP_GET_SEC_DNS ](https://github.com/apache/nuttx-apps/blob/7afa9b51ea97b20ecfabdc8441b518b213d30f76/netutils/pppd/ppp.h#L137C1-L137C1) defines. Then, you will be able to get the DNS servers automatically, and will not have to set the DNS on the `CONFIG_NETDB_DNSSERVER_IPv4ADDR` config. @vshymanskyy 

I'm not sure why these defines were not an explicit option in the menu config (at least I haven't found them there). Anyway, I did some tests here, and it seems that I'm getting the DNS servers properly.