apache/nuttx #7152: Using GPIO with ESP32

SrivamsiMalladi, 2022-09-21:
I am using [Adafruit HUZZAH32 – ESP32 Feather Board](https://www.adafruit.com/product/3405). It has an inbuilt LED connected to pin 13. I am trying to write high and low to it using the [apps/examples/gpio](https://github.com/apache/incubator-nuttx-apps/tree/master/examples/gpio) example, but I can only see `gpio0`, `gpio1`, `gpio2` in `dev`. I tried all three and nothing happens the LED on pin 13. Do they correspond to any other pin? I do not have access to IO0. IO1 or IO2 pins on the board to check them.
I also see that there is [nuttx/arch/xtensa/src/esp32/esp32_gpio.h](https://github.com/apache/incubator-nuttx/blob/master/arch/xtensa/src/esp32/esp32_gpio.h) where the appropriate gpio read and write functions specific to ESP32 are available. But if I use them, I will not be able to use the same application with another board without any changes.

Please let me know if I can use NuttX's generic GPIO driver to read and write (or even use interrupts) with ESP32 or if I am missing something.

acassis, 2022-09-21:
Please look the code at boards/xtensa/esp32/esp32-devkitc/src/esp32_gpio.c it will give you some idea where these /dev/gpioX come from. Basically you will modify the GPIO_OUT1 to point to your LED at pin 13

SrivamsiMalladi, 2022-09-21:
That really helped! Changed GPIO_OUT1 to 13 and the LED is now accessible using the gpio example.
Thanks

closed 2022-09-21 by SrivamsiMalladi