apache/nuttx #7153: Make use of multiplexing (GSM 07.10) protocol for PPPD and GSM modules

vshymanskyy, 2022-09-21:
While the PPPD example works, one needs to exit PPP mode (possibly power-resetting the modem) to send
additional AT commands or read NMEA information from the GSM modems, which proves difficult and requires a lot of task handling.

[GSM 07.10 protocol](https://www.etsi.org/deliver/etsi_ts/101300_101399/101369/07.01.00_60/ts_101369v070100p.pdf) is a suggested way to handle this. It allows multiplexing (AT+CMUX) several logical streams of data over a single uart, i.e. you can have multiple virtual serial ports for:
- AT commands
- PPP
- NMEA (GPS)

acassis, 2022-09-21:
Hi @vshymanskyy AFAIK NuttX doesn't have PPP serial multiplexing, you need to implement the mux protocol. This should be a nice feature to have. If you need help to implement and test it, please let me know.

vshymanskyy, 2022-09-21:
@acassis I'm currently evaluating multiple platforms with PPP support, so I created this issue in process of documenting my findings. I have no immediate need for mux support in NuttX.