apache/nuttx #7163: Netinit bringup for multiple type of network device interfaces

haitomatic, 2022-09-22:
https://github.com/apache/incubator-nuttx-apps/blob/master/netutils/netinit/netinit.c 
According to this, I am wondering how can we bring up all interfaces if we have multiple of them. In my application, I need to enable both ETHERNET and socketCAN at the same time. However, the CONFIG_NET_ETHERNET shadows over CONFIG_NET_CAN and therefore, though registered as a network device, the can0 iface is never brought up. This apparently is not only between socketCAN and Ethernet but also between all other interfaces like wlan, bluetooth,etc.

pkarashchenko, 2022-09-22:
Or you may not boot into NSH, but to custom entry point and use `netlib_ifup()`.

haitomatic, 2022-09-23:
thanks for the comments. Now I see it clear. `rcS` is a good place to bring up all other non-ethernet interfaces :+1:

closed 2022-09-23 by haitomatic