apache/nuttx #7193: Missing SoftAP documentation or feature for ESP32

SrivamsiMalladi, 2022-09-26:
I would to configure ESP32 in SoftAP mode in NuttX, but look like the documentation in the corresponding page ([Wi-Fi SoftrAP](https://nuttx.apache.org/docs/latest/platforms/xtensa/esp32/index.html#wi-fi-softap)) is incomplete.
1. The commands suggested in the documentation of the SoftAP mode section does nothing. 
```
nsh> ifup wlan1
nsh> dhcpd_start wlan1
nsh> wapi psk wlan0 mypasswd 1
nsh> wapi essid wlan1 nuttxap 1`
```
2. Also, the third command `wapi psk wlan0 mypasswd 1` uses `wlan0` while the reset of the commands use `wlan1`. Is that a mistake or it is intentional?
3. Using some external third-party documentations and examples, I enabled the debug messages and got error 19 when trying to use ` dhcpd_start wlan1` which I found means that wlan1 is not found to be started.
```
nsh> dhcpd_start wlan1
dhcpd_openlistener: ERROR: socket failed: 19
dhcpd_run: ERROR: Failed to create socket
```

Is there a way to enable `wlan1`? I tried looking ti

acassis, 2022-09-27:
@SrivamsiMalladi seems the issue is in your side. I tested here and everything is working fine:

```
From NuttX:

nsh> ifconfig
wlan0   Link encap:Ethernet HWaddr ac:67:b2:53:8b:ec at DOWN
        inet addr:0.0.0.0 DRaddr:0.0.0.0 Mask:0.0.0.0

wlan1   Link encap:Ethernet HWaddr ae:67:b2:53:8b:ec at DOWN
        inet addr:0.0.0.0 DRaddr:0.0.0.0 Mask:0.0.0.0

nsh> ifup wlan1
ifup wlan1...OK
nsh> dhcpd_start wlan1
nsh> wapi psk wlan0 mypasswd 1
nsh> wapi essid wlan1 nuttxap 1
nsh> ifconfig
wlan0   Link encap:Ethernet HWaddr ac:67:b2:53:8b:ec at DOWN
        inet addr:0.0.0.0 DRaddr:0.0.0.0 Mask:0.0.0.0

wlan1   Link encap:Ethernet HWaddr ae:67:b2:53:8b:ec at UP
        inet addr:10.0.0.1 DRaddr:10.0.0.1 Mask:255.255.255.0

nsh>

From Linux connected to nuttxap:

alan@dev:~$ ping 10.0.0.1
PING 10.0.0.1 (10.0.0.1) 56(84) bytes of data.
64 bytes from 10.0.0.1: icmp_seq=1 ttl=64 time=3.08 ms
64 bytes from 10.0.0.1: icmp_seq=2 ttl=64 time=3.17 ms
64 bytes from 10.0.0.1: icmp_seq=3 ttl=64 time=3.09 ms
64 bytes from 10.0.0.1: icmp_seq=4 ttl=64 time=4.61 ms
64 bytes from 10.0.0.1: icmp_seq=5 ttl=64 time=2.96 ms
64 bytes from 10.0.0.1: icmp_seq=6 ttl=64 time=1.59 ms
64 bytes from 10.0.0.1: icmp_seq=7 ttl=64 time=8.23 ms
64 bytes from 10.0.0.1: icmp_seq=8 ttl=64 time=1.44 ms
64 bytes from 10.0.0.1: icmp_seq=9 ttl=64 time=1.44 ms
```

acassis, 2022-09-27:
I have the impression that you added a " ` " after the number 1:
```
wapi essid wlan1 nuttxap 1`
```

Please fix it!

SrivamsiMalladi, 2022-09-28:
I am not able to see wlan1 in my ifconfig. What configuration are you using?
This is what I did.
```
$ make distclean
$ ./tools/configure.sh esp32-devkitc:wapi
$ make download ESPTOOL_PORT=/dev/ttyUSB0
$ picocom -b 115200 /dev/ttyUSB0
```

Then, in nsh I did ifconfig.
```
nsh> ifconfig
wlan0   Link encap:Ethernet HWaddr 24:62:ab:d1:ef:30 at UP
        inet addr:10.0.0.2 DRaddr:10.0.0.1 Mask:255.255.255.0

             IPv4   TCP   UDP  ICMP
Received     0000  0000  0000  0000
Dropped      0000  0000  0000  0000
  IPv4        VHL: 0000   Frg: 0000
  Checksum   0000  0000  0000  ----
  TCP         ACK: 0000   SYN: 0000
              RST: 0000  0000
  Type       0000  ----  ----  0000
Sent         0000  0000  0000  0000
  Rexmit     ----  0000  ----  ----
```

And, this is my setup in case it is different from yours
```
nsh> uname -a
NuttX  11.0.0-RC2 72313301e2-dirty Sep 28 2022 11:03:47 xtensa esp32-devkitc
```

> I have the impression that you added a " ` " after the number 1:

The " ` " got added by mistake. Added it for formatting here in GitHub, forgot to remove it. I did not use it in the shell like that.

acassis, 2022-09-29:
No, the wapi board profile is not prepared to SoftAP, you need to use the sta_softap board profile!
You need to see wlan1:
```
nsh> ifconfig
wlan0   Link encap:Ethernet HWaddr ac:67:b2:53:8b:ec at DOWN
        inet addr:0.0.0.0 DRaddr:0.0.0.0 Mask:0.0.0.0

wlan1   Link encap:Ethernet HWaddr ae:67:b2:53:8b:ec at DOWN
        inet addr:0.0.0.0 DRaddr:0.0.0.0 Mask:0.0.0.0

nsh> 
```

SrivamsiMalladi, 2022-09-30:
Oh yea! With sta_softap configuration, I can now see wlan1 in ifconfig. I think this should be clearly mentioned in the documentation.
>It is possible to use ESP32 as an Access Point (SoftAP). Actually there are some boards with a sta_softap which enables this support.

I see this mentioned in the corresponding section, but I think it must be mentioned more clearly to avoid confusion.

So then, I am able to run the following in nsh
```
nsh> ifup wlan1
ifup wlan1...OK
nsh> dhcpd_start wlan1
```
Soon after this command I am able to see nuttxap in the list of available Wi-Fi networks from my Ubuntu system. When I try setting the essid, I get the following error.
```
nsh> wapi essid wlan1 abcdefg 1
esp_wifi_softap_essid: Failed to set Wi-Fi config data ret=12299
ioctl(SIOCSIWESSID): 1
ERROR: Process command (essid) failed.
```

What is it that I am missing again?

acassis, 2022-10-01:
Did you try this way?
```
nsh> ifup wlan1
ifup wlan1...OK
nsh> dhcpd_start wlan1
nsh> wapi psk wlan0 mypasswd 1
nsh> wapi essid wlan1 nuttxap 1
nsh> ifconfig
wlan0   Link encap:Ethernet HWaddr ac:67:b2:53:8b:ec at DOWN
        inet addr:0.0.0.0 DRaddr:0.0.0.0 Mask:0.0.0.0

wlan1   Link encap:Ethernet HWaddr ae:67:b2:53:8b:ec at UP
        inet addr:10.0.0.1 DRaddr:10.0.0.1 Mask:255.255.255.0

nsh>
```

acassis, 2022-10-03:
Also my original "wapi psk wlan0 mypasswd 1" was wrong, it should be "wapi psk wlan1 mypasswd 1" or more correctly "wapi psk wlan1 mypasswd 3" for WPA2. Thanks @tmedicci for noticing the issue!

acassis, 2022-10-03:
@SrivamsiMalladi if you revert this commit https://github.com/apache/incubator-nuttx-apps/commit/be516e4c5e44ac35fa4ab1fed203567f3e98cda6 the SoftAP will work again!

tmedicci, 2022-10-06:
Hello @SrivamsiMalladi, 

I've just submitted a PR that is supposed to fix the issue you are facing. Could you please test it?

https://github.com/apache/incubator-nuttx/pull/7250

SrivamsiMalladi, 2022-10-10:
Hello @tmedicci (@acassis @xiaoxiang781216),
I pulled from the [apache/incubator-nuttx](https://github.com/apache/incubator-nuttx) git repo this morning (so your changes should be there too) and tried to start the SoftAP and it is still not working.
``` 
nsh> ifconfig
wlan0   Link encap:Ethernet HWaddr 24:62:ab:d1:ef:30 at DOWN
        inet addr:0.0.0.0 DRaddr:0.0.0.0 Mask:0.0.0.0

wlan1   Link encap:Ethernet HWaddr 26:62:ab:d1:ef:30 at DOWN
        inet addr:0.0.0.0 DRaddr:0.0.0.0 Mask:0.0.0.0

nsh> dhcpd_start wlan1
nsh> ifconfig
wlan0   Link encap:Ethernet HWaddr 24:62:ab:d1:ef:30 at DOWN
        inet addr:0.0.0.0 DRaddr:0.0.0.0 Mask:0.0.0.0

wlan1   Link encap:Ethernet HWaddr 26:62:ab:d1:ef:30 at UP
        inet addr:10.0.0.1 DRaddr:10.0.0.1 Mask:255.255.255.0
nsh> wapi psk wlan1 mypasswd 3
esp_wifi_softap_auth: Failed to set Wi-Fi config data ret=12299
ERROR: Process command (psk) failed.
nsh> 
```
From the [ESP-IDF error codes documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/error-codes.html), the error code 12299 (0x300B) means that the password is invalid. I tried to set passwords too (like abcdefgh or 12345678 etc.) and everything returns the same error. Despite everything, the AP has already started and I cannot connect to it.

Should I raise this as another issue as this one got closed because of the PR?

tmedicci, 2022-10-10:
> Should I raise this as another issue as this one got closed because of the PR?

I don't think it's necessary. Let's try to fix your issue here ;)

Could you please double-check if my commit is on your repository?

Also, may I ask you if you ran the `make distclean` before building it? It's a necessary step because the `.config` may contain old data. Could you try running:
`make -j distclean && ./tools/configure.sh esp32-devkitc:sta_softap && make -j bootloader && make V=1 flash ESPTOOL_PORT=/dev/ttyUSB0 && minicom -D /dev/ttyUSB0`, please?

I checked with my own board (using this same password) and everything seems to be normal (I didn't have a wireless network with the  `mypasswd` password. It does not check the password/try to connect on this step, so it shouldn't fail there.

SrivamsiMalladi, 2022-10-10:
Yes I have your commit.
```
iotdev@iotdev:~/nuttxspace/nuttx$ git show 5089c5d2fd446197b02c7367e444586aa4587d5e
commit 5089c5d2fd446197b02c7367e444586aa4587d5e
Author: Tiago Medicci Serrano <tiago.medicci@espressif.com>
Date:   Mon Oct 3 17:42:25 2022 -0300

    esp32: fix wapi stack size on board examples
    
    Fix the error on issue 7193:
    (https://github.com/apache/incubator-nuttx/issues/7193)
```
Yes I ran `make distclean` after pulling latest changes and then flashed.

So I did this now
```
$ make distclean
$ ./tools/configure.sh esp32-devkitc:sta_softap
$ make -j distclean && ./tools/configure.sh esp32-devkitc:sta_softap && make -j bootloader && make V=1 flash ESPTOOL_PORT=/dev/ttyUSB0 && minicom -D /dev/ttyUSB0
```
```
nsh> ifconfig
wlan0   Link encap:Ethernet HWaddr 24:62:ab:d1:ef:30 at DOWN
        inet addr:0.0.0.0 DRaddr:0.0.0.0 Mask:0.0.0.0

wlan1   Link encap:Ethernet HWaddr 26:62:ab:d1:ef:30 at DOWN
        inet addr:0.0.0.0 DRaddr:0.0.0.0 Mask:0.0.0.0

nsh> dhcpd_start wlan1
nsh> ifconfig
wlan0   Link encap:Ethernet HWaddr 24:62:ab:d1:ef:30 at DOWN
        inet addr:0.0.0.0 DRaddr:0.0.0.0 Mask:0.0.0.0

wlan1   Link encap:Ethernet HWaddr 26:62:ab:d1:ef:30 at UP
        inet addr:10.0.0.1 DRaddr:10.0.0.1 Mask:255.255.255.0

nsh> wapi psk wlan1 12345678 3
nsh> wapi essid wlan1 mysoftap 1
nsh> 
```

This time I did not get any errors when I tried with this, but the end behaviour is still the same. The AP with the given name gets created, but I cannot connect to it when the password I have specified `12345678`

acassis, 2022-10-10:
@SrivamsiMalladi I think Tiago already discovered your issue: probably you just updated your repository, but still using your old .config that still setting the stack size to less than 4KB!

SrivamsiMalladi, 2022-10-10:
I just now discovered that I can connect to the softap created using the specified password from my Ubuntu machine, but I cannot do the same from my smartphone (Tried with Android). Also I cannot even see the network name in the list of available Wi-Fis in Windows machine.

tmedicci, 2022-10-10:
> ```
> nsh> ifconfig
> wlan0   Link encap:Ethernet HWaddr 24:62:ab:d1:ef:30 at DOWN
>         inet addr:0.0.0.0 DRaddr:0.0.0.0 Mask:0.0.0.0
>
> wlan1   Link encap:Ethernet HWaddr 26:62:ab:d1:ef:30 at [...]

Hi! I'm missing the `ifup wlan1` on your snippet. Did you run that command before those?

I'm able to connect on Android and Linux (manjaro) and Windows. On WIndows it appeared first as a hidden network, but soon the name appeared as expected.

acassis, 2022-10-10:
Here it works fine on desktop e smartphone. Try to turn off and the WiFi on your smartphone. An issue that I faced was that my smartphone silently dropped the connection because it don't have Internet access. So I found an configuration knob to allow to keep connected even when the WiFi doesn't has Internet support.

SrivamsiMalladi, 2022-10-10:
I tried with the `ifup wlan1` now and it is still the same.
On Android, after entering password, I get "Couldn't connect to network" dialogue box.
On Windows, I cannot see hidden network. Even after waiting for a while, nothing.
On Linus, everything works as expected.

tmedicci, 2022-10-11:
Hi @SrivamsiMalladi 

We found that [this commit](https://github.com/apache/incubator-nuttx-apps/commit/be516e4c5e44ac35fa4ab1fed203567f3e98cda6) causes a software regression on our Wi-Fi library. We are working on a proper solution but we have no estimated date to submit it.

As a workaround, please revert this commit locally on your ``apps`` repository. Assuming you are on `ǹuttx`` repo:
```
$ cd ../apps/
$ git revert --no-edit be516e4c5e44ac35fa4ab1fed203567f3e98cda6
``` 
Don't worry, our Wi-FI adapter sets the auth method to WPA2 automatically. 

Regarding your results, please test it again making sure to erase the flash before flashing the new firmware (please adjust your serial port):
```
$ esptool.py -c esp32 -p /dev/ttyUSB0 -b 921600  erase_flash
``` 

Thanks.

SrivamsiMalladi, 2022-10-12:
Hello @tmedicci. Thank you so much for looking into the issue. I tired what you mentioned, but the behavior is still the same.
Cannot connect to the AP from my Android phone.

tmedicci, 2022-10-12:
> Hello @tmedicci. Thank you so much for looking into the issue. I tired what you mentioned, but the behavior is still the same. Cannot connect to the AP from my Android phone.

Hi @SrivamsiMalladi , 

@acassis also has found some difficulty to connect into his Android phone, but it was not related to the device, but to the phone dropping the connection without internet access. Could you, please, try:
1. Check if your phone has some configuration setting to drop connections without internet or to enable mobile data automatically if wi-fi does not provide an internet connection;
2. Setup a different Wi-Fi network on your router/Wi-Fi (without an internet connection and on `10.0.0.1` range) to test if your phone is able to connect directly to the softAP;
3. Try using another Android device to test;

SrivamsiMalladi, 2022-10-18:
Hi @tmedicci,
I am able to connect to other Wi-Fi that do not offer internet. Also I was able to connect to a network with ip 10.42.5.x
When trying to connect to the nuttxap, my phone just prompts "Couldn't connect to network", but when I try to connect to other networks without internet, my phone prompts something like "This network does not offer internet. Do you still want to connect to this network?" But this kind of popup does not appear with nuttxap. So, could it be something else?

SrivamsiMalladi, 2022-10-18:
I also tried changing the ip address from 10.0.0.1 (0x0a000001) to 192.168.1.1 (0xc0a80101) by making the following changes in the `config.h`

```
#define CONFIG_NETUTILS_DHCPD_STARTIP 0xC0A80102
#define CONFIG_NETUTILS_DHCPD_ROUTERIP 0xC0A80101
```

With the above changes, I flashed the sta_softap config and did ifconfig.
```
nsh> ifconfig
wlan0   Link encap:Ethernet HWaddr 24:62:ab:d1:ef:30 at DOWN
        inet addr:0.0.0.0 DRaddr:0.0.0.0 Mask:0.0.0.0

wlan1   Link encap:Ethernet HWaddr 26:62:ab:d1:ef:30 at DOWN
        inet addr:0.0.0.0 DRaddr:0.0.0.0 Mask:0.0.0.0

nsh> dhcpd_start wlan1
nsh> ifconfig
wlan0   Link encap:Ethernet HWaddr 24:62:ab:d1:ef:30 at DOWN
        inet addr:0.0.0.0 DRaddr:0.0.0.0 Mask:0.0.0.0

wlan1   Link encap:Ethernet HWaddr 26:62:ab:d1:ef:30 at UP
        inet addr:192.168.1.1 DRaddr:192.168.1.1 Mask:255.255.255.0
```

Connected to the nuttxap network from Ubuntu system and 192.168.1.2 got assigned to the ubuntu system.
Tried to connect to the nuttxap network from Android smartphone, I get a prompt saying "Couldn't connect to the network".

SrivamsiMalladi, 2022-10-18:
Oh! I just now tried with an Apple device (iPhone) and another android device and they both got connected. 
Probably something is wrong or differently configured with my previous android device like you said @tmedicci.

Sorry for causing the trouble, but I will check and report if I find what it is with my android device.
Closing the issue for now.

closed 2022-10-18 by SrivamsiMalladi

tmedicci, 2022-10-18:
> Oh! I just now tried with an Apple device (iPhone) and another android device and they both got connected. Probably something is wrong or differently configured with my previous android device like [...]

Thank you for your report and, please, let us know if you find the problem regarding your android device.

Best regards,