apache/nuttx #7215: mount file system

xiaotailang, 2022-09-30:
I compile the File system mount example with the config options likes this:
[*] File system mount example  
[*]   Use block device  
(/dev/mtdblock0) Block device name
I can not mount sucess when run mount cmd in console ,and some error info as follow:
nsh> mount                                                                              
mount_main: mounting vfat filesystem at target=/mnt/fs with source=/dev/mtdblock0       
mount_main: mount() returned -1 
i have look the error code it is mean Not owner but i can not understand it, so i need some help.

acassis, 2022-09-30:
Hi @xiaotailang please give more information. We don't know that board you are using, what config you are using, which steps you took enable it, etc. Please describe exactly what you are doing and how you are doing it.

xiaotailang, 2022-10-08:
> Hi @xiaotailang please give more information. We don't know that board you are using, what config you are using, which steps you took to enable it, etc. Please describe exactly what you are doing and how you are doing it.

Okay! My development board is stm32f767, I develop i2c driver under nuttx, my i2c slave device is eeprom, after the development of i2c driver is completed, the i2c device is registered as mtd device, which can be done through at24->bwrite();at24->bread( ) callback functions to read and write eeprom. But when I refer to the method of stm32f103 under nuttx, when I mount eeprom as a block device and initialize the file system as vfat, I find that it cannot be mounted normally. I debug the trace and find that the code logic will check whether there is a boot partition in all partitions of the storage device. , return an error code if there is none, and exit. my code exit from there. So I refer to stm32f103 again to try to initialize eeprom as nxffs file system, and then mount it. After testing, I found that the mount can be successfully and files can be created normally in eeprom. However, I found that whether it is a mount operation or a read and write operation on the eeprom in the file system, the response speed is very slow, about tens of seconds, sometimes even a few minutes. I also did a debug trace and found that no matter what I do on the eeprom, the i2c-driven read and write functions are called thousands of times. I continued to debug and found that the logic related to the nxffs file system will erase the entire storage device and other related operations. These operations will call the read and write functions of the underlying i2c driver, so it is time-consuming to operate eeprom through the file system.
These are the two problems I encountered with the development driver and the file system. Summarized as follows:
1. After i2c is initialized from the device eeprom to the vfat file system, it cannot be mounted normally.
2. The i2c can be mounted normally after the device eeprom is initialized to the nxffs file system, but the response speed is very slow when performing operations such as reading and writing in the file system, including mounting. Almost unusable.
In addition to the configuration required by the i2c driver itself, my configuration reference is mainly the reference stm32f103. A readme is as follows:
CONFIG_I2C=y
CONFIG_MTD=y
CONFIG_MTD_AT24XX=y
CONFIG_AT24XX_SIZE=512
CONFIG_AT24XX_ADDR=0xa0
CONFIG_NSH_ARCHINIT=y
CONFIG_NXFFS=y
 CONFIG_NXFFS_PREALLOCATED=y
CONFIG_STM32_AT24_BLOCKMOUNT=y : Mounts AT24 for NSH
 CONFIG_STM32__AT24_NXFFS=y : Mount the AT24 using NXFFS
  You can then format the AT24 EEPROM for a FAT file system and mount the
  file system at /mnt/at24 using these NSH commands:
   nsh> mkfatfs /dev/mtdblock0
   nsh> mount -t vfat /dev/mtdblock0 /mnt/at24
  Then you an use the FLASH as a normal FAT file system:
   nsh> echo "This is a test" >/mnt/at24/atest.txt
   nsh> ls -l /mnt/at24
   /mnt/at24:
    -rw-rw-rw- 16 atest.txt
   nsh> cat /mnt/at24/atest.txt
   This is a test

Is there something else I need to configure that I haven't considered that is causing the above two problems?

acassis, 2022-10-08:
Hi @xiaotailang you are correct! Using EEPROM with MTD is slow if your EEPROM is I2C. The mount issue you are facing is because the auto-mount command expects that the MTD device be formatted as VFAT before using it. I have a 5 years old tutorial explaining how to use MTD with EEPROM: https://www.youtube.com/watch?v=qzjf2JtgYN0 but that version of NuttX is old.
You could avoid using filesystem and write directly to the EEPROM device file, look at drivers/eeprom/README.txt this is a char device EEPROM driver. I used it in a project storing the packages data directly in the EEPROM. It is away faster.

xiaotailang, 2022-10-09:
> Hi @xiaotailang you are correct! Using EEPROM with MTD is slow if your EEPROM is I2C. The mount issue you are facing is because the auto-mount command expects that the MTD device be formatted as [...]

Thank you very much for taking the time out of your busy schedule to answer my questions and give your valuable advice. I wish you a happy life!

closed 2022-10-19 by acassis