apache/nuttx #7233: mtd/smart: compile error in smart_fsck

SPRESENSE, 2022-10-04:
When `CONFIG_MTD_SMART_FSCK=y`, a compile error is caused by below commit.

9ad75fd95d Added SMART flash filesystem to RP2040

`mtd/smart.c` cannot refer to `SMARTFS_NEXTSECTOR()` and `SMARTFS_SET_NEXTSECTOR()` macro functions.
```
mtd/smart.c: In function 'smart_fsck_file':
mtd/smart.c:5743:19: warning: implicit declaration of function 'SMARTFS_NEXTSECTOR'; did you mean 'SMARTFS_ROOT_DIR_SECTOR'? [-Wimplicit-function-declaration]
 5743 |       logsector = SMARTFS_NEXTSECTOR(chain);
      |                   ^~~~~~~~~~~~~~~~~~
      |                   SMARTFS_ROOT_DIR_SECTOR
mtd/smart.c: In function 'smart_fsck_directory':
mtd/smart.c:5885:11: warning: implicit declaration of function 'SMARTFS_SET_NEXTSECTOR'; did you mean 'SMARTFS_ROOT_DIR_SECTOR'? [-Wimplicit-function-declaration]
 5885 |           SMARTFS_SET_NEXTSECTOR(chain, 0xffff);
      |           ^~~~~~~~~~~~~~~~~~~~~~
      |           SMARTFS_ROOT_DIR_SECTOR
```

pkarashchenko, 2022-10-16:
Is the issue already fixed?

SPRESENSE, 2022-10-19:
Not yet.

closed 2022-10-19 by xiaoxiang781216