apache/nuttx #7271: Build error with SIM_NETUSRSOCK.

fjpanag, 2022-10-10:
When `SIM_NETUSRSOCK` the simulator build fails.

I get the following error:

```sh
/usr/bin/ld: up_usrsock_host.o: in function `usrsock_host_loop':
/path/nuttx/arch/sim/src/sim/posix/up_usrsock_host.c:514: undefined reference to `usrsock_event_callback'
collect2: error: ld returned 1 exit status
```

ethanlcz, 2022-10-10:
> ```shell
> up_usrsock_host.c
> ```

I try to build adb, which enabled SIM_NETUSRSOCK by default and get success result.
Did you upgrade your codebase to latest master?

fjpanag, 2022-10-10:
@ethanlcz I just tried with 6b7fe6ea8e2916f6c06a77c965bcce2d1e53e011 and it still fails.

Can you please share your config, to check for any differences?

xiaoxiang781216, 2022-10-10:
usrsock config is already enabled in daily build:
https://github.com/apache/incubator-nuttx/blob/master/boards/sim/sim/sim/configs/adb/defconfig#L37-L40
Please compare yours with the upper.

fjpanag, 2022-10-11:
`NET_USRSOCK_CUSTOM` is needed for this to build successfully.  
Unfortunately, `SIM_NETUSRSOCK` cannot select `NET_USRSOCK_CUSTOM` since the later is a choice.

I don't think that there is anything more to do on this.

closed 2022-10-11 by fjpanag