apache/nuttx #7274: Undefined behaviour within UBSAN.

fjpanag, 2022-10-10:
It seems that there is undefined behaviour within UBSAN itself.

When it is triggered, it calls itself recursivelly, without limit, exhausting the stack.

The output I get is:
```
[10/10/22 14:25:58] [28] [ EMERG] ubsan_prologue: ================================================================================

[10/10/22 14:25:58] [28] [ EMERG] ubsan_prologue: UBSAN: shift-out-of-bounds in ubsan/ubsan.c:163:35

[10/10/22 14:25:58] [28] [ EMERG] ubsan_prologue: ================================================================================

[10/10/22 14:25:58] [28] [ EMERG] ubsan_prologue: UBSAN: shift-out-of-bounds in ubsan/ubsan.c:163:35

[10/10/22 14:25:58] [28] [ EMERG] ubsan_prologue: ================================================================================

[10/10/22 14:25:58] [28] [ EMERG] ubsan_prologue: UBSAN: shift-out-of-bounds in ubsan/ubsan.c:163:35

[10/10/22 14:25:58] [28] [ EMERG] ubsan_prologue: ================================================================================

[10/10/22 14:25:58] [28] [ EMERG] ubsan_prologue: UBSAN: shift-out-of-bounds in ubsan/ubsan.c:163:35

[10/10/22 14:25:58] [28] [ EMERG] ubsan_prologue: ================================================================================
```

*(And repeats several thousand times...)*

fjpanag, 2022-10-10:
Forgot to mention, this is on simulator.

fjpanag, 2022-10-10:
Hmm this is very odd.

By reverting specifically 32a21a1b6762081e2f925bc63b9daf61f5c8b15a, the problem goes away...  
I get no UBSan warnings.

(Of course using the exact same flags: `-fsanitize=undefined`)

xiaoxiang781216, 2022-10-10:
The only difference is that the new change pass "-fsanitize=undefined", not -fsanitize=undefined. Does this make the difference?

fjpanag, 2022-10-10:
> The only difference is that the new change pass "-fsanitize=undefined", not -fsanitize=undefined. Does this make the difference?

Yes, you are correct.

When I use:

```
ifeq ($(CONFIG_SIM_UBSAN),y)
  ARCHOPTIMIZATION += -fsanitize=undefined
else ifeq ($(CONFIG_MM_UBSAN_ALL),y)
  ARCHOPTIMIZATION += -fsanitize=undefined
endif
```

I get no UBSan warnings at all.

When I use:

```
ifeq ($(CONFIG_SIM_UBSAN),y)
  ARCHOPTIMIZATION += "-fsanitize=undefined"
else ifeq ($(CONFIG_MM_UBSAN_ALL),y)
  ARCHOPTIMIZATION += "-fsanitize=undefined"
endif
```

I get some warnings, both for UBSan and my code.

So it seems like in the first case UBSan is not actually working.

Anyway, the original issue remains: UBSan fails itself.

xiaoxiang781216, 2022-10-10:
Could you provide a patch to strip out quote from option? Or wait @no1wudi take a look tomorrow.

fjpanag, 2022-10-10:
No, the correct version seems to be **with** the quotes.

xiaoxiang781216, 2022-10-10:
Ok, anyway @no1wudi can take a look tomorrow.

fjpanag, 2022-10-10:
> Ok, anyway @no1wudi can take a look tomorrow.

Sure! Just adding my findings, in the hope that I will help.

The offending line seems to be:
```
return ((int64_t)ulong_val) << extra_bits >> extra_bits;
```

I printed the value of `extra_bits` and it is 0 at the time of the problem.

So the following is UB?
```
return ((int64_t)ulong_val) << 0 >> 0;
```

fjpanag, 2022-10-10:
> So the following is UB?

Answering my own question, I think it is indeed UB, because `ulong_val` is cast to a signed type.  

Changing it to the following seems to resolve the issue, but I would appreciate any input from @no1wudi .

```c
      return (int64_t)(ulong_val << extra_bits >> extra_bits);
```

closed 2022-10-12 by xiaoxiang781216