apache/nuttx #7300: How to get IPv4 address with usrsock ioctl?

anhnt-dev, 2022-10-13:
With device uses usrsock, how can I get the IPv4 address from ioctl:
``` C
int netlib_get_ipv4addr(FAR const char *ifname, FAR struct in_addr *addr)
{
  ...
          struct ifreq req;
          strlcpy(req.ifr_name, ifname, IFNAMSIZ);
          ret = ioctl(sockfd, SIOCGIFADDR, (unsigned long)&req);
  ...
  return ret;
}
```
I checked in usrsock_host_ioctl() function, there's no argument was referenced to **_req_**  variable.

xiaoxiang781216, 2022-10-13:
sim's usrsock isn't a full implementation, you can reference one implementation here:
https://github.com/apache/incubator-nuttx-apps/blob/master/netutils/usrsock_rpmsg/usrsock_rpmsg_server.c#L745-L787

anhnt-dev, 2022-10-13:
> 

Thanks @xiaoxiang781216 , a good point to start.

xiaoxiang781216, 2022-10-13:
Let's close it now. If you have more question, please reopen this issue.

closed 2022-10-13 by xiaoxiang781216