apache/nuttx #7307: System does not start on STM32 when CONFIG_STACK_CANARIES is enabled.

fjpanag, 2022-10-13:
When I enable the option `CONFIG_STACK_CANARIES` is not running.

It builds correctly, I flash it to the board, but `main()` is never entered. Somehow this option breaks the build.

This is affecting some arch'es but not all.  
I have tried this option on simulator where everything runs fine. Specifically on STM32 however there is a problem.

---

Things I tried so far:

I used my debugger to see what is going on when the system hangs.  
The program always hangs in function `stm32_rcc_enablelsi()`, file stm32_lsi.c, specifically at line 64.  
The code cannot exit the loop.

I tried to remove this loop (or make it exit with my debugger), but the code immediately halts to the next similar loop.  
It is not the specific check that causes the issue.

I disassembled the code both when the option is enabled or disabled.  
The assembly is identical, no wrong instructions there.

I followed the assembly with my debugger, and everything seemed fine.

I checked the register address (`STM32_RCC_CSR`) that is loaded on the CPU register, and it is identical in both cases, and correct.

Just after, the CPU loads the contents of the register `STM32_RCC_CSR` to a CPU register.  
The following values are read:
* Without stack canaries: 0x14000003
* With stack canaries: 0x14000000

Due to a reason that my brain cannot comprehend, this option somehow affects how the CPU reads the hardware registers, or something like that. I have spend a fair amount of time on this, and I cannot understand the issue. I am out of ideas now.

*(A hardware issue is completely ruled out.)*

acassis, 2022-10-18:
I tested enabling CONFIG_STACK_CANARIES for ESP32 and worked fine. Seems to be really specific for STM32. I will try later on STM32F4Discovery board to confirm

fjpanag, 2022-12-22:
I experience a very similar issue on an LPC1769. I see that somewhere inside `__start()` an unexpected IRQ is happening, and the system panics.

This is only happening when `CONFIG_STACK_CANARIES` is enabled.

pkarashchenko, 2022-12-24:
I just tried `CONFIG_STACK_CANARIES=y` with `same70-qmtech:nsh` board and it seems to be working. Will try with STM32 later

pkarashchenko, 2022-12-24:
I just checked `CONFIG_STACK_CANARIES=y` with `STM32F407VE` based board and it reached NSH console without any issues. Maybe there are some other options that give side effect in combination with `CONFIG_STACK_CANARIES`?