apache/nuttx #7331: Question: How to enable nsh over composite usb configuration using RP2040

abdallah-ali-abdallah, 2022-10-16:
Hello, I'm using raspberry pi pico (RP2040) and trying to build an application requires exposing 4 USB composite devices
for example, I need to run the following over USB2.0 interface:

1- NSH 
2- serial control interface 1
3- serial control interface 2 
4- pass-through SD card using MSC composite instance

Examples usbmsc, usbnsh and composite works perfectly but I can't find the correct mix of configurations that enables me to start nsh as composite instance.

Thank you.

acassis, 2022-10-18:
Hi @abdallah-ali-abdallah I quickly look at defconfig from composite example and it shows that is enables the mass storage and serial interface (CDC/ACM). In your case you will need to extend the arm/rp2040/common/src/rp2040_composite.c to support your configuration. First you will need to verify if the rp2040's USB have enough end-points available and study a little bit to understand how composite works. Also look the example for other boards, it could help.

abdallah-ali-abdallah, 2022-10-19:
Hi @acassis 

Thank you for your help, I did what you explained and after some hacking, tweaking and a lot of jumping between files I was able to operate the NSH over composite mode and used multiple serial instances for extra data transfers

I will state all the steps I did (and I think there might be a way less hacky, please if you have any recommendation for better way mention it) 
------------ steps goes as following --------

1- I started with **tools/configure.sh raspberry-pico:composite**

2- Disable the default serial consol over **UART0** 
**_Device Drivers → Serial Driver Support → No serial console_**
[image]

3- Activate alternative consol **_Application Configuration → NSH Library → Console Configuration_**
[image]

4- Now for testing I added 2 serial CDC/ACM by editing **arm/rp2040/common/src/rp2040_composite.c**
[image]

5- (Here is the tricky part) When I tried to compile the project, no usb detected, after some jumping around files, I missed a simple fact, NSH need access to /dev/ttyACM0 which never initialized because by default we need to call application "conn" from NSH (which not started yet...uuuh).

So the fix is to init the USB composite before starting the NSH, which I did by modifying **nsh_initialize** inside **_apps/nshlib/nsh_init.c_**
where I added 2 actions applied to USB 

-   ctrl.action   = BOARDIOC_USBDEV_INITIALIZE;
-   ctrl.action   = BOARDIOC_USBDEV_CONNECT;

[image]

6- Recompile everything and upload ...

Finally,, dmesg on linux telling me I have 2 serial interfaces attached to my machine
[image]

and NSH is working perfectly on /dev/ttyACM0

... extra problem.

Based on documentation [here](https://nuttx.apache.org/docs/latest/applications/nsh/config.html)

[image]

NSH will not print '\r\n' to devices except /dev/consol. that causes the consol to read shofted and miss aligned lines

I applied some fix to this by modifying **nuttx/drivers/serial/serial.c** and forced serial to convert every '\n' to '\r\n'
(based on if TERMIOS is enabled or not)

This
[image]
Or that
[image]

Now everything works perfectly and NSH lines printed correctly on minicom.

So, Do you have any recommendation for **_cleaner way_** to do it without modifying any core nuttx files for manipulating the nsh_initialize function 

Thank you.

acassis, 2022-10-19:
Hi @abdallah-ali-abdallah that is amazing!!! Kudos!
Could you please submit this test as a guide to be included at https://nuttx.apache.org/docs/latest/guides/index.html ?

You can do it cloning this repo and adding your guide as Documentation/guides/composite.rst

Thank you very much!

acassis, 2022-10-19:
Yes, I have a solution to fix the issue you are facing! You don't need to enable the NSH_ALTCONDEV, instead just enable the CONFIG_CDCACM_CONSOLE ! Look boards/arm/rp2040/raspberrypi-pico/configs/usbnsh/defconfig ;-)

abdallah-ali-abdallah, 2022-10-20:
> 

Tried that, but it didn't work, even the composite device didn't initiate any instance.
The only configuration worked very fine if the following.

[image]

and 

[image]
 
I will try to test the same configurations with another board like STM32

abdallah-ali-abdallah, 2022-10-20:
> Hi @abdallah-ali-abdallah that is amazing!!! Kudos! Could you please submit this test as a guide to be included at https://nuttx.apache.org/docs/latest/guides/index.html ?
> 
> You can do it cloning this repo and adding your guide as Documentation/guides/composite.rst
> 
> Thank you very much!

Greate, I will do more testing with different boards to make sure it will work on multiple architectures. Then I will add it to documentation.

Thank you for your advice :))

acassis, 2022-10-20:
@xiaoxiang781216 @pkarashchenko did you see this '\n' issue before?

abdallah-ali-abdallah, 2022-10-20:
> @abdallah-ali-abdallah what serial console terminal are you using? Does it support VT102?

I use minicom mainly which support VT102, Also tested miniterm.py and same result in all cases
Without modifying the **if condition** for '\n' -> '\r\n' in **serial.c** all lines are miss aligned.

pkarashchenko, 2022-10-21:
> @xiaoxiang781216 @pkarashchenko did you see this '\n' issue before?

I will need to look around, but endlines are tricky with terminals and many configuration options are usually present in the terminal program itself. I think I'm using a serial over pyserial python module and that also has options to pad '\r' or use RAW stream with no padding.

acassis, 2022-10-22:
@abdallah-ali-abdallah @pkarashchenko reading it again: "NSH will not print '\r\n' to devices except /dev/consol. that causes the consol to read shofted and miss aligned lines"...

I think this is expected because you could use serial devices with NSH that are not suppose to need "\r\n". Remember: Unix never needed "\r\n" because "\n" always mean "move to new line" (at the begin of a new line). I think "\n" meaning move the new line at current column position is something created with DOS (or CP/M ?). It exist on NuttX because some serial console terminal expect the DOS/Windows behavior.

@abdallah-ali-abdallah with your modification everything someone send a "\n" char in the serial it will become a "\r\n" and could prevent some devices to work. We need to figure-out why CONFIG_CDCACM_CONSOLE is not working...

I found a useful information at apps/nshlib/Kconfig
```
                Normally, the serial console device is a UART and RS-232
                interface.  However, if USBDEV is defined, then a USB
                serial device may, instead, be used if the one of
                the following are defined:

                PL2303 and PL2303_CONSOLE - Set up the Prolifics PL2303
                emulation as a console device at /dev/console.

                CDCACM and CDCACM_CONSOLE - Set up the CDC/ACM serial
                device as a console device at dev/console.

                NSH_USBCONSOLE and NSH_USBCONDEV - Sets up some other USB
                serial device as the NSH console (not necessarily dev/console).
```

No idea why CONFIG_NSH_USBCONSOLE couldn't become a /dev/console there should be some reason why Greg decided to do this way.

acassis, 2022-10-22:
@abdallah-ali-abdallah I have a suggestion: keep CONFIG_NSH_USBCONSOLE enabled and don't define the CONFIG_NSH_USBCONDEV, it should put use /dev/console as USB Console:

```
/* The default USB serial console device */

#  ifndef CONFIG_NSH_USBCONDEV
#    if defined(CONFIG_CDCACM)
#      define CONFIG_NSH_USBCONDEV "/dev/ttyACM0"
#    elif defined(CONFIG_PL2303)
#      define CONFIG_NSH_USBCONDEV "/dev/ttyUSB0"
#    else
#      define CONFIG_NSH_USBCONDEV "/dev/console"
#    endif
#  endif
```

acassis, 2022-10-28:
@abdallah-ali-abdallah did you get everything working?

abdallah-ali-abdallah, 2022-10-29:
> @abdallah-ali-abdallah did you get everything working?

Not yet, got busy in work, but I found out some more issues.

for some reason creating more than 2 virtual ACM makes the device super slow. it get so slow that it could take 2 ~ 3 mintues to show / load all /dev/ttyACM* instance (tested on ubuntu and Fedora) and will run the same test on opensuse to check if it's OS dependent issue

acassis, 2022-11-01:
Hi @abdallah-ali-abdallah do you know if rp2040 has OpenOCD support? My suggestion is to use GDB+OpenOCD to debug it. I suspect there is some semaphore or equivalent thing interfering in the performance. Also check the amount of USB end-point, maybe you hit the limit and it is waiting from some free end-points to continue

abdallah-ali-abdallah, 2022-11-01:
> @abdallah-ali-abdallah I have a suggestion: keep CONFIG_NSH_USBCONSOLE enabled and don't define the CONFIG_NSH_USBCONDEV, it should put use /dev/console as USB Console:
>
> ```
> /* The default USB [...]

So I tried that also, Nothing changes, same behavior no '\r' included in end of line