apache/nuttx #7366: how to add microkernelized characteristics on nuttx 8.2

yuening1999, 2022-10-20:
Hi, all 
Could anyone give some advices about the topic? for instance network component，file system and device driver ,etc.
I investigate the relevant code to network on nuttx 8.2, I found that all net protocol APIs(e.g., arp,icmp) were called directly from other components (e.g., device driver),which is a tight-coupling style. how to implement a little microkernelized characteristics?

acassis, 2022-10-20:
@yuening1999 do you need to use this ancient version of NuttX? Maybe using some recent version will be easier. I think to use the network in separated you will need some kind of IPC. What MCU are you using? Does it support Protected Mode? or only Flat mode?

yuening1999, 2022-10-21:
@acassis Do you mean that nuttx 10.x or 11 had done some microkernelized characteristics ? In additon, I have saw some of implementation and usage for message queue(mq_xxx) on nuttx 8.2, why do you think I need some kind of IPC using the network in separation on nuttx 8.2, I have no understand the point. If current latest version of nuttx had made some microkernelized characteristics, could you tell what to do and how to do.  ?

Oh, forgot, my MCU support protected build, armv8m core.

acassis, 2022-10-22:
Hi @yuening1999 I didn't explain it correctly: using a recent NuttX version (ie. 11.x) is better because it has significant improvements related to the network. We don't have microkernel implemented on recent NuttX version. Are you integrating seL4 on NuttX or just isolating NuttX kernel to the bare minimum to become a microkernel?
BTW: NuttX 8.2 doesn't have armv8m core support, NuttX 11.0 has it!

yuening1999, 2022-10-23:
@acassis  Thank your for the detailed explaination. my team currently ready to port/add some microkernel architecture to Nuttx 8.2 by means of referencing to Phoenix-RTOS implementation,  Meanwhile, seL4 is also a choice,  @but we still do not make a decison.

BTW, we has implement protected mode on new MCU made in China, it work fine, although it  currently do not support armv8m

acassis, 2022-10-24:
Hi @yuening1999 really cool!!! Please send the normal support to this new chip to NuttX mainline, it will help more people to discover about this new chip. I think the microkernel support also should be something that could be useful to have on NuttX mailing, but it will depend on user selecting BSD License support instead of Apache only.

acassis, 2022-10-28:
@yuening1999 may we close this issue?

closed 2022-10-29 by yuening1999