apache/nuttx #7367: usrsock: wget using gs2200m fails by usrsock update

SPRESENSE, 2022-10-20:
webclient_perfom used from wget application returns an error with -118 (EHOSTUNREACH).
Not only gs2200m but other usrsock daemons have the same problem.

If I revert the following commits, it works well.
c752536 usrsock:fix blocking socket connect operation hang issue
be06d00 usrsock: update state to connected when recv sento ready under connecting

It seems that it is caused by changing the handling of `USRSOCK_EVENT_SENDTO_READY`,
but I do not know the details. Can anyone support this issue?

Thanks.

### How to reproduce
```
$ ./tools/configure.sh spresense:wifi
$ make menuconfig

I have changed some configurations to check my operation.

CONFIG_EXAMPLES_WGET=y                          // Enable examples/wget
CONFIG_NETDB_DNSSERVER_IPv4ADDR=0x08080808      // Use DNS server 8.8.8.8
CONFIG_WIFI_BOARD_IS110B_HARDWARE_VERSION_10C=y // Select my hardware type
CONFIG_WL_GS2200M_DISABLE_DHCPC=n               // Enable DHCP
```
```
nsh> gs2200m <AP name> <Password> &
gs2200m [5:50]
nsh>
nsh> ifconfig
wlan0   Link encap:Ethernet HWaddr 14:5a:fc:f8:16:f6 at UP
        inet addr:192.168.1.18 DRaddr:192.168.1.1 Mask:255.255.255.0

nsh> wget http://www.example.com/index.html
webclient_perform failed with -118
```

xiaoxiang781216, 2022-10-20:
@zhhyu7 could you take a look?

zhhyu7, 2022-10-20:
> @zhhyu7 could you take a look?

Ok, let me check this issue.

zhhyu7, 2022-10-20:
@SPRESENSE you can verify with below patch:
https://github.com/apache/incubator-nuttx-apps/pull/1366

SPRESENSE, 2022-10-20:
@zhhyu7 Thank you. I could confirm that this issue has been resolved.
@masayuki2009 could you check an above patch?

masayuki2009, 2022-10-20:
@zhhyu7 @SPRESENSE 
I confirmed that the PR fixes the issue.

closed 2022-10-20 by masayuki2009