apache/nuttx #7370: ostest failling on second execution

tmedicci, 2022-10-20:
After @masayuki2009 added new tests (thanks!), `ostest` is failling on second execution, on `task_restart` test:

```
user_main: task_restart test

Test task_restart()
restart_main: setenv(VarName, VarValue, TRUE)
restart_main: Started restart_main at PID=119
restart_main: Started with argc=4
restart_main: argv[0]="ostest"
restart_main: argv[1]="This is argument 1"
restart_main: argv[2]="Argument 2 here"
restart_main: argv[3]="Lastly, the 3rd argument"
restart_main: Variable=VarName has value=VarValue
restart_main:  ERROR: task_restart failed
ostest_main: Exiting with status 256
```
As it's related to OS functionalities, I thought this issue should be reported here (instead `apps` repo).

**How to reproduce**
Just set the `sim:ostest` defconfig, build and run:
`make -j distclean && ./tools/configure.sh -l sim:ostest && make -j && ./nuttx`

tmedicci, 2022-10-20:
> > After @masayuki2009 added new tests (thanks!), ostest is failling on second execution, on task_restart test:
> 
> @tmedicci I did not add new tests but added ASSERT(false) in [apache/incubator-nuttx-apps#1351](https://github.com/apache/incubator-nuttx-apps/pull/1351). So the problem would be in `task_restart`

Yes, I meant that! (I'm sorry I wasn't that clear). Thank you for adding these asserts. The problem on `task_restart` possibly already existed, but it wasn't being checked.

masayuki2009, 2022-10-20:
>The problem on task_restart possibly already existed, but it wasn't being checked.

@tmedicci 
I think so.
Anyway, thank you for reporting the issue.

masayuki2009, 2022-10-20:
@tmedicci 
I've just fixed the issue.

closed 2022-10-20 by tmedicci