apache/nuttx #7393: Problems with priority inheritance.

fjpanag, 2022-10-22:
Priority inheritance seems broken again.

After the recent fixes, I run ostest on my board, and worked correctly.  
I decided to enable `CONFIG_PRIORITY_INHERITANCE` again.

Since then, I occasionally get a failed assertion in `sem_holder.c`, line 105.  
The message is: `"ERROR: Insufficient pre-allocated holders"`.

This is however impossible in my case.  
I have set `CONFIG_SEM_PREALLOCHOLDERS` to 8, but at the time of the failure 8 tasks do not even exist in my system, let alone wait for a semaphore!

Somehow the list of the holders is miss-treated, and the assertion fires erroneously.

fjpanag, 2022-10-22:
I spent some time with this, and I noticed something.  
It may or may not be important.

Here are two stack traces, of the only 2 tasks that were actually doing something during the issue:

[image]

[image]

Notice:

The `gui` task is the failing task.  
It fails within a call to `nxsem_wait()`.

The second task is the `init` task.  
This has been scheduled out in favor of `gui` **during** the release of a semaphore.

It has called `up_reprioritize_rtr()` that caused the second thread to execute.  
**But**, going back in the stack trace I see a previous call to `nxsem_post()`. This function is supposedly run with the scheduler locked (i.e. `sched_lock()`), and there is this nice comment:

```c
      /* Don't let any unblocked tasks run until we complete any priority
       * restoration steps.  Interrupts are disabled, but we do not want
       * the head of the ready-to-run list to be modified yet.
       *
       * NOTE: If this sched_lock is called from an interrupt handler, it
       * will do nothing.
       */
```

The code runs under the assumption that no other task can run, but it does!  
The internal restoration of the priority causes a context switch.

Maybe this has to be converted to a critical section?  
Or maybe the flow needs to be changed?

fjpanag, 2022-10-22:
Hmm, there is a critical section protecting all these, but `nxsched_running_setpriority()` does not respect it.  
It does a context switch nevertheless.

acassis, 2022-10-22:
Good finding @fjpanag !!! Did you bisect to find which PR broke it?

xiaoxiang781216, 2022-10-22:
I guess this patch: https://github.com/apache/incubator-nuttx/pull/6318

fjpanag, 2022-10-22:
> Good finding @fjpanag !!! Did you bisect to find which PR broke it?

I don't think bisecting will help here, because priority inheritance saw some work recently. It was already broken before the fixes, so I don't think testing an earlier revision will provide any meaningful results.

pkarashchenko, 2022-10-22:
@fjpanag what ARCH are you using and what is the `.config`?
According to next code
```
#ifndef CONFIG_SMP
bool nxsched_add_readytorun(FAR struct tcb_s *btcb)
{
  FAR struct tcb_s *rtcb = this_task();
  bool ret;

  /* Check if pre-emption is disabled for the current running task and if
   * the new ready-to-run task would cause the current running task to be
   * pre-empted.  NOTE that IRQs disabled implies that pre-emption is
   * also disabled.
   */

  if (rtcb->lockcount > 0 && rtcb->sched_priority < btcb->sched_priority)
    {
      /* Yes.  Preemption would occur!  Add the new ready-to-run task to the
       * g_pendingtasks task list for now.
       */

      nxsched_add_prioritized(btcb, &g_pendingtasks);
      btcb->task_state = TSTATE_TASK_PENDING;
      ret = false;
    }
```
the `up_reprioritize_rtr` should not cause the context switch if current task locked the preemption. Are you using `CONFIG_SMP=y`?

pkarashchenko, 2022-10-22:
But maybe the problem is not in `nxsched_add_readytorun `, but in `nxsched_remove_readytorun` since code in `up_reprioritize_rtr` is doing
```
      /* Remove the tcb task from the ready-to-run list.
       * nxsched_remove_readytorun will return true if we just
       * remove the head of the ready to run list.
       */

      switch_needed = nxsched_remove_readytorun(tcb);

      /* Setup up the new task priority */

      tcb->sched_priority = (uint8_t)priority;

      /* Return the task to the specified blocked task list.
       * nxsched_add_readytorun will return true if the task was
       * added to the new list.  We will need to perform a context
       * switch only if the EXCLUSIVE or of the two calls is non-zero
       * (i.e., one and only one the calls changes the head of the
       * ready-to-run list).
       */

      switch_needed ^= nxsched_add_readytorun(tcb);

      /* Now, perform the context switch if one is needed */

      if (switch_needed)
```
so if `nxsched_remove_readytorun` returned `true` and `nxsched_add_readytorun` returned `false` then the context switch will happen.

I mean that in special situation when we are having only 2 tasks and for example 1 task in in pending list and another in ready to run. Then we are trying to modify the priority of a ready to run that is currently a `this_task()` -> `(FAR struct tcb_s *)g_readytorun.head`, so we are removing it from read to run list and try to add it back. But in this case code in `nxsched_add_readytorun`:
```
  FAR struct tcb_s *rtcb = this_task();
  bool ret;

  /* Check if pre-emption is disabled for the current running task and if
   * the new ready-to-run task would cause the current running task to be
   * pre-empted.  NOTE that IRQs disabled implies that pre-emption is
   * also disabled.
   */

  if (rtcb->lockcount > 0 && rtcb->sched_priority < btcb->sched_priority)
```
should be a `NULL` pointer dereference since `rtcb` will be `NULL` at this point after `nxsched_remove_readytorun` is executed. Anyway it would be good to get some more info to investigate.

fjpanag, 2022-10-26:
@pkarashchenko This issue is observed on an STM32F427. OF course, `CONFIG_SMP` is not defined.

pkarashchenko, 2022-10-26:
I have SMT32F407 based board. Maybe you can share the defconfig, so I can reproduce and debug the issue?

acassis, 2022-10-26:
@fjpanag you can save your defconfig (make savedefconfig) and put it in pastebin.com for example

pkarashchenko, 2022-10-28:
Ok. Thank you. I will try to replicate the issue on my end. Meanwhile could you please try running ostest with priority inheritance on your setup, just to align on system behavior.

pkarashchenko, 2022-10-28:
@fjpanag I've drafted some changes in https://github.com/apache/incubator-nuttx/pull/7464 but most probably will not be able to try it out till Monday.

pkarashchenko, 2022-12-14:
@fjpanag I renewed https://github.com/apache/nuttx/pull/7464 in non-SMP case. Could you please try this changes in your case?

davids5, 2023-02-17:
@fjpanag - any thing further to report testing of #7464 ?

fjpanag, 2023-02-17:
@davids5 @pkarashchenko sorry for taking long to respond.

The thing is that with my current builds I have a very hard time reproducing the issue **without** the patch, so I cannot evaluate whether the fix actually fixed the root cause.

Nevertheless, I will re-test the fix, to ensure that at least it seems OK. If the original issue re-appears, I will report back.

Please give me a couple of days to re-test #7464. I expect to finish within the weekend.

fjpanag, 2023-02-19:
Bad news...

I am now able to reproduce the original issue consistently.

I tried on latest master, and indeed the assertion fails.

Then, I cherry-picked the changes (7a125ae37ca162ac1aa8803550adfa26e83a0ba9), but the failure remains identical.

Unfortunately, #7464 does neither fix the problem, nor affect it in any way.

fjpanag, 2023-04-05:
@pkarashchenko you have drafted some changes in #7464, that didn't solve the issue.

Do you think that this is a good change nevertheless (and thus merge it)?  
Or this fix becomes obsolete with the latest changes in priority inheritance (and thus maybe close the PR)?

I will spend more time on this issue when I find some free time, but in the meantime #7464 can move forward independently, I guess...

pkarashchenko, 2023-04-05:
Yes. I need to spend more time on that change and create ostest for that specific issue.

fjpanag, 2023-04-05:
> Yes. I need to spend more time on that change and create ostest for that specific issue.

Thank you. If you thing that your fix may affect this issue in any way, please let me know, so I can re-test it.  
Since I can reproduce it easily...

pkarashchenko, 2023-04-05:
@fjpanag just one question. Is the trace remaining the same with and without https://github.com/apache/nuttx/pull/7464 ? I mean, do you still see `up_reprioritize_rtr` in the back trace for one of the threads while another is trying to wait on a semaphore?

pkarashchenko, 2023-04-06:
It is more, can you please actualize the back trace logs because there was some renaming done from the original issue reporting.

fjpanag, 2023-04-06:
> It is more, can you please actualize the back trace logs because there was some renaming done from the original issue reporting.

What do you mean? [These](https://github.com/apache/nuttx/issues/7393#issuecomment-1287783155) traces are provided verbatim, I haven't edited anything.

pkarashchenko, 2023-04-06:
Yes, but in the latest master there is no `up_reprioritize_rtr` function. It has been renamed to `nxsched_reprioritize_rtr`. So I'm wondering how the latest back trace look like.

fjpanag, 2023-04-06:
> Yes, but in the latest master there is no `up_reprioritize_rtr` function. It has been renamed to `nxsched_reprioritize_rtr`. So I'm wondering how the latest back trace look like.

Ah OK, I understand. I will do it soon.

pkarashchenko, 2023-04-06:
@fjpanag I've looked into `sched_lock` again and made some changes in https://github.com/apache/nuttx/pull/7464
Please try it out and provide your feedback

fjpanag, 2023-04-07:
@pkarashchenko I have some trouble reproducing the issue with the latest revisions.
It happens sometimes, but not in the same way, the traces are not clear at all.

I am bisecting to find how/when things changed, but it takes time...

---

After hundreds of tests over the last couple of days on this matter, I am now very convinced that #7488 is highly related, or even the same bug.

Unfortunately #7488 is not reproducible on the latest revisions, but I believe it is not fixed (rather, it is just not triggered).  
I tried bisecting, but I couldn't make any sense as to what caused the change in behaviour.

In the case of #7488, I am almost sure that it is a case of double-freeing the holder, I just haven't found yet the exact point in the code.

I will keep working on this.

fjpanag, 2023-06-06:
I've been using and testing priority inheritance after the recent changes, and everything seems OK.

I haven't witnessed any other issues.

I think this can be closed for the moment.

closed 2023-06-06 by fjpanag