apache/nuttx #7435: Assertion fails within STM32 MAC driver.

fjpanag, 2022-10-26:
I am using Ethernet on an STM32F427, and I am plagued with an assertion failing constantly.

It is on file `stm32_eth.c`, line 1372.:  
`DEBUGASSERT(dev->d_len == 0 && dev->d_buf == NULL);`

Checking with my debugger I see that `dev->d_len` has a non zero value (which however seems sane, not garbage).  
`dev->d_buf` is NULL, as it ought to be.

I had set `CONFIG_NET_ETH_PKTSIZE` to 640, and this problem appeared quite often.  
Now, I tried to set this config to 1514 and to my surprise, the problem vanished. I haven't seen this assertion failing again.

Maybe the problem is solved, or maybe it is now so scarce that I haven't noticed it?  

I am thinking of two cases here:
* Either there is something "special" with the value 1514, and the driver does not work with smaller values.
* Or having a small buffer triggers the execution of some buggy code (for example by dropping large packets).

I guess that if it is the second case, the problem is still there but it is just temporarily masked for me.

acassis, 2022-10-26:
@fjpanag I suggest you to use JTAG/SWD + GDB to compare what is different in both cases. It will be help to identify the issue faster!

fjpanag, 2023-02-20:
So, I used my firmware with `CONFIG_NET_ETH_PKTSIZE` set to 1514 over the couple last months, and I didn't see the assertion failing again.

Today, as an experiment, I re-set it back to 640. Using the latest master.  
After about an hour of operation, the same assertion failed, in the exact same way.

I think I can conclude pretty safely that the STM32 driver has problems with this setting.

fjpanag, 2023-02-21:
@davids5 can you please tell me what is the value of `CONFIG_NET_ETH_PKTSIZE` that you are using?

davids5, 2023-02-21:
> @davids5 can you please tell me what is the value of `CONFIG_NET_ETH_PKTSIZE` that you are using?

https://github.com/PX4/PX4-Autopilot/search?q=CONFIG_NET_ETH_PKTSIZE

1518

Did you see https://github.com/PX4/NuttX/pull/137 ?

fjpanag, 2023-03-04:
> > @davids5 can you please tell me what is the value of `CONFIG_NET_ETH_PKTSIZE` that you are using?
> 
> https://github.com/PX4/PX4-Autopilot/search?q=CONFIG_NET_ETH_PKTSIZE
> 
> 1518
> 
> Did you see [PX4/NuttX#137](https://github.com/PX4/NuttX/pull/137) ?

Thank you, I was not aware of this issue.  
It seems that, at least for the moment, a solution to this is to set appropriately `CONFIG_NET_ETH_PKTSIZE`.

However, there is still something I find quite odd.  
Since #2985, when a small `CONFIG_NET_ETH_PKTSIZE` is selected, the larger packets should be dropped. I see [here](https://github.com/apache/nuttx/pull/2985#issuecomment-791512546) that this is confirmed.

However, now I see that the assertion fails (instead of silently dropping the unsupported packet).  
Possibly something has changed over the years, and now the logic in #2985 is not working correctly?

fjpanag, 2023-03-04:
Hmm, I haven't understood the code completely yet, but when `stm32_freesegment()` is called shouldn't the fields `d_len` & `d_buf` be invalidated too?

I think that `stm32_freesegment()` correctly drops the oversized packet, but `d_len` remains on the last value, which will result in the assertion to fail (as it expects everything reset to 0).

fjpanag, 2023-03-04:
Yep, these are definitely related.  
@davids5 could you please have a look?

As a test, I set `CONFIG_NET_ETH_PKTSIZE` to a lower value.  
I then send a UDP packet to the device, with size greater than the Ethernet packet size.

The code added in #2985 correctly triggers, and drops the packet.  
Some times everything works correctly. But repeating this 3-4 times will result in a system crash.

Testing my idea, I added `dev->d_len = 0;` just after `stm32_freesegment(priv, rxcurr, priv->segments);` and the issue vanished!

I think that some kind of handling of `dev->d_len` must be added indeed.

davids5, 2023-03-06:
@fjpanag I do not have a setup at the moment. Has the usage of dev->d_len, changed in resent Network changes? Is the assert still valid? If it is and we are not resetting dev->d_len = 0 on a drop could that be the cause?