apache/nuttx #7475: RaspberryPi-Pico-W is not getting WiFi

acassis, 2022-10-30:
Hi @curuvar someone told me that raspberrypi-pico-w was not working with recent kernel, then I tested to confirm:

-------------------------------------------------------------------------
SDK Setup:

$ cd /home/alan

$ git clone -b 1.1.2 https://github.com/raspberrypi/pico-sdk.git

$ export PICO_SDK_PATH=/home/alan/pico-sdk

-------------------------------------------------------------------------

NuttX Conf. / Compilation

$ cd nuttxspace/nuttx

$ ./tools/configure.sh raspberrypi-pico-w:telnet

$ make menuconfig

Application Configuration  --->
   Network Utilities  --->
       -*- Network initialization  --->
           WAPI Configuration  --->
               (PutHereYourRouterName) SSID
               (PutHereYourRouterPassword) Passprhase

$ make -j

Press and hold BOOTSEL button, then plug the USB Cable

Copy nuttx.uf2 para RPI-RP2 disk

--------------------------------------------------------------------------

NuttX Testing:

nsh> uname -a
NuttX  10.4.0 99cfffc96a-dirty Oct 30 2022 10:30:55 arm raspberrypi-pico-w
nsh> ifconfig
lo      Link encap:Local Loopback at RUNNING
        inet addr:127.0.0.1 DRaddr:127.0.0.1 Mask:255.0.0.0

wlan0   Link encap:Ethernet HWaddr 00:00:00:00:00:00 at DOWN
        inet addr:0.0.0.0 DRaddr:10.0.0.1 Mask:255.255.255.0

nsh> ifup wlan0
ifup wlan0...Failed
nsh> renew wlan0
ERROR: dhcpc_request() failed
nsh>

TA1DB, 2023-01-18:
Any news guys?

acassis, 2023-01-18:
Still failing here! I enabled the debug network and got more info about the error:

```
nsh> ifconfig
lo      Link encap:Local Loopback at RUNNING
        inet addr:127.0.0.1 DRaddr:127.0.0.1 Mask:255.0.0.0

wlan0   Link encap:Ethernet HWaddr 00:00:00:00:00:00 at DOWN
        inet addr:0.0.0.0 DRaddr:0.0.0.0 Mask:255.255.255.0

nsh> wapi psk wlan0 MyPassword 3
wpa_driver_wext_process_auth_param: ERROR: SIOCSIWAUTH(param 0 value 0x4) failed: 1)
ERROR: Process command (psk) failed.
```

It will need more investigation from people using this board...

acassis, 2023-01-18:
@PeterBee97 Do I need to have the "Infineon 43439 firmware file" to get the WiFi working? There is not documentation about it in the README.tx file

PeterBee97, 2023-01-19:
> @PeterBee97 Do I need to have the "Infineon 43439 firmware file" to get the WiFi working? There is not documentation about it in the README.tx file

Sorry Alan, I only have the old Pico without W and haven't worked with WiFi on this board. @anchao Any idea?

anchao, 2023-01-19:
I do not have pico W on hand, just placed an order, I will fix this issue later.

anchao, 2023-02-06:
@acassis @TA1DB please help to review PR https://github.com/apache/nuttx/pull/8439 , which fixed CYW43 not working properly on pico-W

anchao, 2023-02-06:
Please note that the setup process of the cyw43 driver has changed:

```
1. Download Raspberry Pi Pico SDK and update submodule(cyw43-driver)

  $ git clone -b 1.4.0 https://github.com/raspberrypi/pico-sdk.git
  $ cd pico-sdk
  $ git submodule update --init --recursive lib/cyw43-driver
...
```

closed 2023-02-06 by xiaoxiang781216

linguini1, 2024-10-01:
I am encountering this issue again using the 2.0.0 pico-sdk. The wireless driver appears to be the same version: `lib/cyw43-driver/firmware/43439A0-7.95.49.00.combined`

I am using the telnet configuration as well.

```console
nsh> wapi psk wlan0 "myPassword" 3
[ 9] netdev_ifr_ioctl: cmd: 35634
[ 9] wpa_driver_wext_process_auth_param: ERROR: SIOCSIWAUTH(param 0 value 0x4) failed: 1)
nsh>
nsh> ifup wlan0
[ 3] netdev_ifr_ioctl: cmd: 1818
[ 3] bcmf_wl_active: Entered
ifup wlan0...Failed
```

UAV-Pilot, 2024-10-05:
I encountered the same error with pico sdk 2.0. Performed "git submodule update --init --recursive lib/cyw43-driver".

I set SSID and passphrase in menuconfig, and also tried to set them in nsh.

nsh> renew wlan0
ERROR: netlib_obtain_ipv4addr() failed
nsh> ifconfig wlan0
wlan0   Link encap:Ethernet HWaddr 00:00:00:00:00:00 at DOWN mtu 5nsh>
nsh>
nsh> ifup  wlan0
ifup wlan0...Failed
 
 ChatGPT suggested to set mac address:
         ifconfig wlan0 hw ether XX:XX:XX:XX:XX:XX
 but got error: 
         nsh: ifconfig: argument invalid
         
nsh> wapi show wlan0
wlan0 Configuration:
       IP: 0.0.0.0
  NetMask: 255.255.255.0
nsh> wapi scan wlan0
nsh> wapi scan_results wlan0
nsh> ifup wlan0
ifup wlan0...Failed

May @anchao or someone else please investigate this issue again?

UAV-Pilot, 2024-10-05:
Since this issue is closed, I submitted a new issue: https://github.com/apache/nuttx/issues/13828 with regarding to Pi Pico SDK 2.0