apache/nuttx #7488: Kasan error when exiting task.

fjpanag, 2022-10-31:
I just got a KASan error, when a task is exited.  
I am using the simulator for this test.

It seems related to the semaphores.

I may not be able to further troubleshoot this today, but here is a stack trace in case anyone has any ideas:

[image]

xiaoxiang781216, 2022-10-31:
Can you try CONFIG_SIM_ASAN which could report more detailed info

fjpanag, 2022-11-09:
The issue is within `nxsem_freeholder()` were there is a NULL-pointer dereference.  
I see that `pholder->htcb` is NULL, but the code tries to use `pholder->htcb->holdsem` directly, without a check.

I am not sure if a check is missing here, or indeed this field shouldn't be NULL (and thus the root cause is elsewhere).

@pkarashchenko This issue may be of interest to you.

pkarashchenko, 2022-11-09:
Yes. I will try to take a look into it as soon as I can. Most probably this weekend

pkarashchenko, 2022-11-09:
I did a brief look and see the 3 calls of `nxsem_freeholder` are done from `nxsem_foreachholder` loop that has `DEBUGASSERT(pholder->htcb != NULL);` and the only other call of `nxsem_freeholder` is from `nxsem_release_all` that is called from `nxsem_recover`. Based on your log it is a case of `nxsem_recover` that is called from `nxtask_recover`. We need to debug how it happens that `htcb->holdsem->htcb != htcb` and is actually `NULL`.
I'm filling like `nxsem_recover` has a bug. Do you have any sample code that can help me to reproduce an issue?

fjpanag, 2022-11-10:
@pkarashchenko Unfortunately I haven't been able to create a minimum working example.

My code runs Lua in a pthread, and executes the Lua tests suite (which is quite complex, and exercises lots of APIs).  
After exiting this pthread, the error is triggered.

I am trying to narrow it down more...

fjpanag, 2023-04-11:
I finally got it!

The semaphore is freed before its holder is released!

Later on when the holder is to be released, as it still has a reference to the (now deallocated) semaphore, KASAN throws an error.

*I am currently testing on 59416af3e697ab89a85f8598751654422f63d46c, as it is one of the revisions that reproduce the issue easily.*

So, what I did is to add the following to `mm_free()`:

```c
  extern dq_queue_t g_readytorun;
  struct tcb_s *rtcb = g_readytorun.head;
  if (rtcb->holdsem && rtcb->holdsem->sem == mem)
  {
	  DEBUGASSERT(0);
  }
```

This is to indicate me that a semaphore is to be freed, while it is still being referenced by this task's holder.  
The assertion indeed fires, and KASAN is right to complain later on.

The stack trace at the time of the failure is this:

[image]

To my understanding the issue is that the file is locked at fs_tmpfs.c line 1418, and then deallocated at line 1446 without it being unlocked first.

And thus, the lock causes a holder to be allocated for this task which ends up pointing to a destroyed object.

fjpanag, 2023-04-11:
This is fixed in 8c52633 (so that's why it is not reproducible on the latest master).

Closing...

closed 2023-04-11 by fjpanag