apache/nuttx #7508: NuttX TCP/IP handshake is failing

acassis, 2022-11-02:
During the 3-way handshake the server refuses to accept NuttX connection:

```
1    0.000000000    192.168.1.100    8.8.8.8    DNS    72    Standard query 0x1ea4 A mqtt.tago.io
2    0.038974140    8.8.8.8    192.168.1.100    DNS    104    Standard query response 0x1ea4 A mqtt.tago.io A 99.83.172.119 A 75.2.83.130
3    0.040266049    192.168.1.100    99.83.172.119    TCP    60    4098 → 1883 [SYN] Seq=0 Win=5488 Len=0 MSS=536
4    0.189729115    99.83.172.119    192.168.1.100    TCP    58    1883 → 4098 [SYN, ACK] Seq=0 Ack=0 Win=65535 Len=0 MSS=0
5    0.190332031    192.168.1.100    99.83.172.119    TCP    60    4098 → 1883 [ACK] Seq=1 Ack=1 Win=5488 Len=0
```

After many hours debugging and sniffing the network I figure out that the issue is happening because NuttX always starts with TCP sequence number equal 0. According to https://datatracker.ietf.org/doc/html/rfc1948 the initial value should be random:

```
   The initial sequence numbers are intended to be more or less random.
   More precisely, RFC 793 specifies that the 32-bit counter be
   incremented by 1 in the low-order position about every 4
   microseconds.  Instead, Berkeley-derived kernels increment it by a
   constant every second, and by another constant for each new
   connection.  Thus, if you open a connection to a machine, you know to
   a very high degree of confidence what sequence number it will use for
   its next connection.  And therein lies the attack.
```

xiaoxiang781216, 2022-11-02:
@acassis https://github.com/apache/incubator-nuttx/pull/6154 randomize port number, you can apply the similar approach for the sequence number.

acassis, 2022-11-02:
Perfect! Thank you @xiaoxiang781216 ! I will do it and submit a PR

closed 2022-11-13 by xiaoxiang781216