apache/nuttx #7523: Do we need -funwind-tables option for some boards?

raiden00pl, 2022-11-04:
Do we still need `-funwind-tables` option for some boards ? It's not compatible with CLANG compiler so we need additional condition in Make.defs.

This option was introduced in 4d43b05db3bdd79f5438798d3f05625b746e71f6 for stm32f4discovery and then probably copied to  other boards based on stm32f4discovery. 
@acassis do you remember the details of the problem (it was a long time ago, PR probably on the old bitbucket repo) ? 

Support for stack unwidner is already supported by CONFIG_UNWINDER_ARM option:

https://github.com/apache/incubator-nuttx/blob/d4ba93067ee76e656c39d42e6dcf5f8daa48322e/arch/arm/src/common/Toolchain.defs#L77-L79

but I'm not sure if this has any effects on the C++ code with exceptions enabled.

Boards with this option in Make.defs:

> [raiden00:~/git/RTOS/nuttx/nuttx/boards]$ grep -R unwind-tabl                                                                                                                                                                                                         
arm/stm32/b-g431b-esc1/scripts/Make.defs:ARCHCFLAGS += -funwind-tables
arm/stm32/b-g431b-esc1/scripts/Make.defs:ARCHCXXFLAGS += -funwind-tables
arm/stm32/nucleo-g431rb/scripts/Make.defs:ARCHCFLAGS += -funwind-tables
arm/stm32/nucleo-g431rb/scripts/Make.defs:ARCHCXXFLAGS += -funwind-tables
arm/stm32/b-g474e-dpow1/scripts/Make.defs:ARCHCFLAGS += -funwind-tables
arm/stm32/b-g474e-dpow1/scripts/Make.defs:ARCHCXXFLAGS += -funwind-tables
arm/stm32/nucleo-g474re/scripts/Make.defs:ARCHCFLAGS += -funwind-tables
arm/stm32/nucleo-g474re/scripts/Make.defs:ARCHCXXFLAGS += -funwind-tables
arm/stm32/stm32f4discovery/scripts/Make.defs:#   ARCHCFLAGS += -funwind-tables
arm/stm32/stm32f4discovery/scripts/Make.defs:#   ARCHCXXFLAGS += -funwind-tables
arm/stm32/nucleo-g431kb/scripts/Make.defs:ARCHCFLAGS += -funwind-tables
arm/stm32/nucleo-g431kb/scripts/Make.defs:ARCHCXXFLAGS += -funwind-tables
arm/eoss3/quickfeather/scripts/Make.defs:  ARCHCFLAGS += -funwind-tables
arm/eoss3/quickfeather/scripts/Make.defs:  ARCHCXXFLAGS += -funwind-tables

acassis, 2022-11-04:
I found it on my email:

"I was facing this error caused by recent modifications on mainline ldscripts:

...
LD: nuttx
arm-none-eabi-ld: section .data LMA
[000000000800e644,000000000800e6ab] overlaps section .ARM.exidx LMA
[000000000800e640,000000000800e647]
make[1]: *** [Makefile:185: nuttx] Error 1

Searching for this error I found a solution here:
https://stackoverflow.com/questions/30778614/arm-none-eabi-ld-section-arm-exidx-overlaps-section-data

The inclusion of "-funwind-tables" fixed the error, but we need to
doable check if it will not generate size effects for other people
(same way previous modification generated side effects to me)."

acassis, 2022-11-04:
The thread name on Google group is:
PATCH: Fix .ARM.exidx section overlap

acassis, 2022-11-04:
We need to check again and depending on result we can put CONFIG_UNWINDER_ARM as dependence for these boards

closed 2022-11-05 by xiaoxiang781216