apache/nuttx #7549: UART Break detection - ESP32

SrivamsiMalladi, 2022-11-07:
I am using UART with NuttX on ESP32. I am looking for an interrupt that fires when a break is detected like it is done with [`UART_BREAK` event in ESP-IDF ](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/uart.html#_CPPv417uart_event_type_t). Is something like this available in NuttX as I could not find it in the documentation? I tried my best to look it up in the source code and did not find it.

acassis, 2022-11-08:
@SrivamsiMalladi on NuttX (as on Unix, Linux) it is implemented with termios **BRKINT**, but AFAIK none uart driver on NuttX has support to it. You need to modify the ESP32 driver to see if BRKINT is requested and setup the uart controller to generated an interruption that will be used to send a signal using termios. What is your need to this UART Break, some specific protocol?

SrivamsiMalladi, 2022-11-08:
@acassis Yea I have a specific protocol followed in my bus that has break in the start of every frame of data. I did a work around by checking if no bytes were received in the last x ms/us, then just assume that it is a break byte and it worked in my case.

Also, I do not read the 0x00 from UART when I receive a break byte. So is the break detected internally and already handled somewhere in the Serial Device Driver? How do I check it?

With the interrupts, I did not understand how to use interrupts with UART which I think is due to the lack of documentation!?
The [documentation for Timer Driver](https://nuttx.apache.org/docs/latest/components/drivers/character/timer.html) is really good. Something of that sort for [Serial Device Driver](https://nuttx.apache.org/docs/latest/components/drivers/character/serial.html) would be great.

Please let me know if I am missing something? Like should I go through the general implementation of Serial drivers on Linux and the same thing could be used for NuttX? Sorry for asking (simple?) questions. I am just used to embedded software programming and learning RTOSs lately. I do not have experience with Linux or its drivers.

acassis, 2022-11-08:
Hi @SrivamsiMalladi I totally agree with you, we need to improve our documentation to cover things like the UART driver, TERMIOS, etc.

Yes, the BREAK is detected by the ESP32 UART controller. Please take a look at UART_BRK_DET_INT on ESP32 TRM. Basically you need to enable it and when the ISR was fired it need to inform the OS that it happened.

I suggest you to ask in the mailing list when is the right way to do it, probably it will use some signal to let the kernel to advertise the application about this event.