apache/nuttx #7598: mpu protected mode debug

xiaotailang, 2022-11-15:
Under mpu protected mode, I added a new system call, But appeared a panic , and I tried to debug, but when using gdb tracing, only the code in kernel mode can be traced, and the code in application mode cannot be traced. So I want to know in mpu protected mode, how does gdb track the code of application state and kernel state at the same time.

patacongo, 2022-11-15:
Can't you just add the symbol table for the user space to GDB using the add-symbol-file command?

superxyxiao, 2022-11-16:
Is there any document about adding new system calls under the protection of mpu? I add a new system call in the following file, but when debugging, I found that the address of the new system call in the g_stublookup array is 0.

path/syscall/syscall_lookup.h:SYSCALL_LOOKUP(mycall,                      0, STUB_mycall)               
path/syscall/syscall_stublookup.c:uintptr_t STUB_mycall(int nbr);
path/syscall/syscall.csv:"mycall","nuttx/mycall.h","","int"

cat path/include/nuttx/mycall.h 

#ifndef __ASSEMBLY__
#undef EXTERN
#if defined(__cplusplus)
#define EXTERN extern "C"
extern "C"
{
#else
#define EXTERN extern
#endif

int mycall(void);

#undef EXTERN
#ifdef __cplusplus
}
#endif

path/include/sys/syscall.h
#define SYS_mycall                    (SYS_prctl + 1)
#define SYS_maxsyscall                (SYS_prctl + 2)

cat path/sched/mycall/mycall.c 

#include <nuttx/compiler.h>
#include <nuttx/mycall.h>
int mycall(void)
{
  int num=0;
  num=num+2;
  return num;
}

superxyxiao, 2022-11-16:
I use  version is  nuttx-8.2

xiaoxiang781216, 2022-12-07:
@superxyxiao nuttx-8.2 is a very old release(2019/11), why not use the last dev branch?