apache/nuttx #7617: Exception generated when gdb debugging mpu protected mode on stm32f767 development board

superxyxiao, 2022-11-18:
This is what I do as follows：
(gdb) load nuttx
Loading section .text, size 0x1523f lma 0x8000000
Loading section .ARM.exidx, size 0x8 lma 0x8015240
Loading section .data, size 0x64 lma 0x8015248
Start address 0x8000268, load size 86699
Transfer rate: 27 KB/sec, 10837 bytes/write.
(gdb) load nuttx_user.elf 
Loading section .userspace, size 0x34 lma 0x8020000
Loading section .text, size 0xacd3 lma 0x8020034
Loading section .ARM.exidx, size 0x8 lma 0x802ad08
Loading section .data, size 0x8 lma 0x802ad10
Start address 0x80200bc, load size 44311
Transfer rate: 13 KB/sec, 7385 bytes/write.
(gdb) symbol-file nuttx
Reading symbols from nuttx...done.
(gdb) add-symbol-file nuttx_user.elf  0x8020034 
add symbol table from file "nuttx_user.elf" at
	.text_addr = 0x8020034
(y or n) y
Reading symbols from nuttx_user.elf...done.
(gdb) hd hello_main
Undefined command: "hd".  Try "help".
(gdb) hbreak  hello_main
Hardware assisted breakpoint 1 at 0x802989e: file hello_main.c, line 53.
(gdb) c
Continuing.

Program received signal SIGTRAP, Trace/breakpoint trap.
exception_common () at armv7-m/gnu/up_exception.S:142
142		mrs		r0, ipsr				/* R0=exception number */
(gdb)

acassis, 2022-11-18:
Hi @superxyxiao try to put the breakpoint at __start and them you will be able to track what is causing the interruption.

superxyxiao, 2022-11-22:
Hi @acassis Generate the same error,The serial terminal is also unable to input.  by the way ,I use 8.2 version . my opertion whether error？

(gdb) target remote :4242
Remote debugging using :4242
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
0x08002b18 in ?? ()
(gdb) load nuttx
Loading section .text, size 0x1523f lma 0x8000000
Loading section .ARM.exidx, size 0x8 lma 0x8015240
Loading section .data, size 0x64 lma 0x8015248
Start address 0x8000268, load size 86699
Transfer rate: 27 KB/sec, 10837 bytes/write.
(gdb) load nuttx_user.elf 
Loading section .userspace, size 0x34 lma 0x8020000
Loading section .text, size 0xacd3 lma 0x8020034
Loading section .ARM.exidx, size 0x8 lma 0x802ad08
Loading section .data, size 0x8 lma 0x802ad10
Start address 0x80200bc, load size 44311
Transfer rate: 13 KB/sec, 7385 bytes/write.
(gdb) symbol-file nuttx
Reading symbols from nuttx...done.
(gdb) add-symbol-file nuttx_user.elf 0x8020034
add symbol table from file "nuttx_user.elf" at
	.text_addr = 0x8020034
(y or n) y
Reading symbols from nuttx_user.elf...done.
(gdb) hbreak __start 
Hardware assisted breakpoint 1 at 0x800026e: file chip/stm32_start.c, line 328.
(gdb) c
Continuing.

Program received signal SIGTRAP, Trace/breakpoint trap.
exception_common () at armv7-m/gnu/up_exception.S:142
142		mrs		r0, ipsr				/* R0=exception number */
(gdb)