apache/nuttx #7670: GigaDevice 4 byte address status register bit - incorrect?

TimJTi, 2022-11-23:
in file drivers/mtd/gd25.c we see at line 109:

#define GD25_SR1_EN4B               (1 << 3)  /* Bit 3: Enable 4byte address */

Checking the data sheets for 2 or 3 devices in this family I see that it is actually Bit 0.

Before I create a PR for a fix, I thought I'd best check in case there's a device in the family where it actually *is* Bit 3, meaning the fix is more than just changing that one line.

acassis, 2022-11-24:
@xiaoxiang781216 why I can't mark "@wangyanjiong" to take a look on it?

acassis, 2022-11-24:
@TimJTi I noticed that the Status Register of many GD25 devices are different. So probably Bit 3 is correct for some devices. Many devices are using Bit 11 instead of Bit 3 or Bit 0. So this bit is device dependent (or family device dependent).

xiaoxiang781216, 2022-11-24:
> @xiaoxiang781216 why I can't mark "@WangYanjiong" to take a look on it?

Because he doesn't develop NuttX for a long time. But we just use one particular GD25 flash, we don't have more information than you and @TimJTi :(. I think the best source is raise question to GigaDevice FAE, or read all GD25 variant datasheet to find the difference.

TimJTi, 2022-11-24:
I have checked a number of GigaDevice datasheets and it seems that only those in the GD25Q family (JEDEC CHIP IDENT 0x40) have the 4 byte read status in Bit 3 and the other types have it in bit 0 as per existing code.

I have made changes and will submit a PR.

closed 2022-11-25 by acassis