apache/nuttx #7731: SAMA5D2 has no support for Flexcom SPI - best method?

TimJTi, 2022-11-29:
To add flexcom spi support for SAMA5D2 I can:

1) create dedicated **sam_flexcom_spi.c** and .h files, keeping it separate from the existing **sam_spi.c** code. Or
2) Enhance the **sam_spi.c** code to  add the flexcom spi structs etc,. and the related set up functions

Any preference from anyone?

acassis, 2022-11-30:
@TimJTi I don't know about SAMA5D2, but I think SAM5's flexcom is similar to SAMD21's sercom. Take a look how it is done on sam_sercom.c and how sam_spi.c uses it.

TimJTi, 2022-11-30:
Aha! I searched the repo for flex and found no other helpful references and didn't know they were called sercon in other devices.

I'll go and take a look. It's why I asked 😎

TimJTi, 2022-11-30:
SAMD21 and SAME5 **only** have sercom ports which can be SPI/I2C/USART/LIN whereas SAMA5D2 has 2x dedicated SPI, 2x dedicated I2C, 2x dedicated UART, **plus* 5xFLEXCOM which are quite possibly unique to the SAMA5D2. Trust me to pick this one.

Since the arch is currently set up to manage the main SPI, I2C (TWI) and UARTs independently to the FLEXCOM ports and there was skeleton code for FLEXCOM USART (now finished off by me) I will stick with that philosophy and keep FLEXCOM SPI completely separate.

I have no FLEXCOM I2C (TWI) devices on my board so I'll leave that for someone else, but will be adding LIN support sometime next year.

PR to follow in a few days...followed by the usual 17+ iterations before I get it right...

acassis, 2022-12-01:
Yes, I think you can take inspiration from sercom driver to implement a separated FLEXCOM SPI driver. You need to think how to let users select it in a flexible way in the menuconfig. So user could select 1 or 2 dedicated SPI and up to 5 FLEXCOM SPI. Since you are not planing to add FLEXCOM I2C (and UART?) just create the base file to be extended in the future case someone want to add them.

TimJTi, 2022-12-01:
USART support already there and now working (via one of my PRs), so this will add SPI via a new sam_flexcom_spi.c file.

Kconfig support is there already, actually, but just doesn't do anything for SPI as no underlying support. Each of the FLEXCOMs have their own bool settings for SPI, TWI or USART and, in time, will gain one each for LIN.

I will proceed. Thanks

adamfeuer, 2022-12-21:
Hi @TimJTi I saw that SPI Flexcom support got accepted in https://github.com/apache/nuttx/commit/e6bfaa14ee353bdcc90682ceeb2d2ee456dd9b4a - it is without DMA. Do you plan to add DMA support?

Also are you working on I2C Flexcom support? (I am working on SAMA5D2 drivers too and don't want to duplicate work.)

TimJTi, 2022-12-21:
I will be looking at DMA but not for a while - so if you do it first you'll save me the hassle! I think the SAMA5D2 DMA is fundamentally not working in NuttX so it does need some work.

I won't be adding I2C as my custom board has no need of it, meaning I have nothing to test it with of course.

TimJTi, 2022-12-21:
Oh - I also have ClassD audio as a WIP at the moment and will be adding that in due course.

adamfeuer, 2022-12-21:
Ok, then I'll see if I can get the SPI Flexcom DMA working. I need that for my application. I'll will be adding I2C Flexcom support too. I'll also test the SPI Flexcom code with my custom board.

adamfeuer, 2022-12-21:
Thank you for adding the SPI Flexcom, by the way - it will save *me* a lot of hassle too. :)

BTW what is your application if you can say? If you would rather email, my contact info is on my GitHub profile. 

Here's mine: https://github.com/starcat-io/jupiter-nano