apache/nuttx #7734: `pthread_mutex_lock(3)` must not return `EINTR`

midokura-xavi92, 2022-11-30:
According to the POSIX.1-2017 [`man` page for `pthread_mutex_lock(3)`](https://pubs.opengroup.org/onlinepubs/9699919799/functions/pthread_mutex_lock.html):

> These functions shall not return an error code of [EINTR].

While it seems this was taken into account for the [implementation for `pthread_mutex_lock(3)`](https://github.com/apache/nuttx/blob/4281acdb837d4d60d384a3d6a612edc7df309c90/libs/libc/pthread/pthread_mutex_lock.c#L71-L72), it seems there is a possible code path that might return `EINTR`:

- `pthread_mutex_lock(3)` [calls](https://github.com/apache/nuttx/blob/4281acdb837d4d60d384a3d6a612edc7df309c90/libs/libc/pthread/pthread_mutex_lock.c#L89) `pthread_mutex_timedlock(3)`.
    - `ret` [is assigned](https://github.com/apache/nuttx/blob/4281acdb837d4d60d384a3d6a612edc7df309c90/sched/pthread/pthread_mutextimedlock.c#L190) to the return value from `pthread_mutex_take(3)`.
        - `ret` [is assigned](https://github.com/apache/nuttx/blob/4281acdb837d4d60d384a3d6a612edc7df309c90/sched/pthread/pthread_mutex.c#L169) to the return value from `pthread_sem_take(3)`.
            - `ret` [is assigned](https://github.com/apache/nuttx/blob/4281acdb837d4d60d384a3d6a612edc7df309c90/sched/pthread/pthread_initialize.c#L72) to the return value from `nxsem_wait`. Please take into account `abs_timeout` [was assigned to `NULL`](https://github.com/apache/nuttx/blob/4281acdb837d4d60d384a3d6a612edc7df309c90/libs/libc/pthread/pthread_mutex_lock.c#L89) by `pthread_mutex_lock(3)`.
                - `ret` [is assigned](https://github.com/apache/nuttx/blob/4281acdb837d4d60d384a3d6a612edc7df309c90/sched/semaphore/sem_wait.c#L201) to `rtcb->errcode`, which, according to [the documentation](https://github.com/apache/nuttx/blob/4281acdb837d4d60d384a3d6a612edc7df309c90/sched/semaphore/sem_wait.c#L191-L194), could be `EINTR`.

Could you please confirm this potentially invalid behaviour from `pthread_mutex_lock(3)`?

acassis, 2022-11-30:
@midokura-xavi92 do you have simple example of code to trigger the return of this invalid value?

midokura-xavi92, 2022-11-30:
I do not have one, but this issue was reported internally by @TachibanaRyota, so maybe they can provide further details.

casaroli, 2022-12-01:
Maybe the change needed is in this line:

https://github.com/apache/nuttx/blob/4281acdb837d4d60d384a3d6a612edc7df309c90/sched/pthread/pthread_mutextimedlock.c#L190

So that we ignore EINTR and comply with POSIX in `pthread_mutex_timedlock
` 
https://github.com/apache/nuttx/blob/4281acdb837d4d60d384a3d6a612edc7df309c90/sched/pthread/pthread_initialize.c#L55

xiaoxiang781216, 2022-12-07:
@casaroli could you create a patch for this?

casaroli, 2022-12-07:
https://github.com/apache/nuttx/pull/7810

closed 2022-12-07 by xiaoxiang781216