apache/nuttx #7795: An error occurred building maix-bit:nsh

GooTal, 2022-12-06:
With `nuttx` and `apps` both in **master** branch(commit id `af2b4914208cad676d64823f5e0bf08871617d80`) , i built nuttx image using command:
```
./tools/configure.sh -l maix-bit:nsh
make 
```
And an error occurred:
[image]

I checkout to releases/11.0 branch and built image with the same command. No error occurred.

I\`m not quite familiar with nuttx. Any idea how to fix this? Thanks.

xiaoxiang781216, 2022-12-06:
could you try this version:
https://github.com/apache/nuttx/blob/master/tools/ci/docker/linux/Dockerfile#L180

GooTal, 2022-12-06:
Thanks, I\`ll try this later. 
But it seems that riscv cross cross compile tools is not the root cause. The branch `release/11.0` can be successfully built using riscv64-unknown-elf-gcc (12.1.0 on my ubuntu 22.04).

xiaoxiang781216, 2022-12-06:
But maix-bit/nsh pass the daily ci https://github.com/apache/nuttx/actions/runs/3622886642/jobs/6108177030:
```
====================================================================================
Configuration/Tool: maix-bit/nsh
------------------------------------------------------------------------------------
  Cleaning...
  Configuring...
  Building NuttX...
  Normalize maix-bit/nsh
```
So, it should the environment difference.

xiaoxiang781216, 2022-12-06:
Look like it's better to update the toolchain used by ci to the last version and fix the compiler issue.

xiaoxiang781216, 2022-12-07:
Let's close this issue if you have more comment, please reopen again.

closed 2022-12-07 by xiaoxiang781216