apache/nuttx #7802: Network performance deterioration and weird behaviour on ESP32

lucasssvaz, 2022-12-06:
After the changes introduced by #7616, the ESP stopped responding to pings when manually configuring the wifi parameters (through `wapi` commands). This does not happen when the parameters are set through the `menuconfig`.

Also, throughput of `iperf` seems to be severly affected:
```
nsh> iperf -s -p 11111 -i 1
       IP: 192.168.1.73

 mode=tcp-server sip=192.168.1.73:11111,dip=0.0.0.0:5001, interval=1, time=0
accept: 192.168.1.245,44420

           Interval         Transfer         Bandwidth

   0.00-   1.01 sec          0 Bytes    0.00 Mbits/sec
   1.01-   2.02 sec          0 Bytes    0.00 Mbits/sec
   2.02-   3.03 sec          0 Bytes    0.00 Mbits/sec
   3.03-   4.04 sec         60 Bytes    0.00 Mbits/sec
   4.04-   5.05 sec         60 Bytes    0.00 Mbits/sec
   5.05-   6.06 sec         60 Bytes    0.00 Mbits/sec
   6.06-   7.07 sec         60 Bytes    0.00 Mbits/sec
   7.07-   8.08 sec         60 Bytes    0.00 Mbits/sec
   8.08-   9.09 sec         60 Bytes    0.00 Mbits/sec
   9.09-  10.10 sec         60 Bytes    0.00 Mbits/sec
  10.10-  11.11 sec       8820 Bytes    0.06 Mbits/sec
  11.11-  12.12 sec      10280 Bytes    0.01 Mbits/sec
  12.12-  13.13 sec      10280 Bytes    0.00 Mbits/sec
  13.13-  14.14 sec      20500 Bytes    0.08 Mbits/sec
  14.14-  15.15 sec      33640 Bytes    0.10 Mbits/sec
  15.15-  16.16 sec      38020 Bytes    0.03 Mbits/sec
  16.16-  17.17 sec      39480 Bytes    0.01 Mbits/sec
  17.17-  18.18 sec      39480 Bytes    0.00 Mbits/sec
  18.18-  19.19 sec      57000 Bytes    0.13 Mbits/sec
  19.19-  20.20 sec      57000 Bytes    0.00 Mbits/sec
  20.20-  21.21 sec      61380 Bytes    0.03 Mbits/sec
  21.21-  22.22 sec      64300 Bytes    0.02 Mbits/sec
  22.22-  23.23 sec      64300 Bytes    0.00 Mbits/sec
  23.23-  24.24 sec      64300 Bytes    0.00 Mbits/sec
  24.24-  25.25 sec      64300 Bytes    0.00 Mbits/sec
```

We (me, @tmedicci, @acassis and @gustavonihei) are still investigating the issue and will add here any relevant information.

acassis, 2022-12-06:
Some important information:
If the WiFi/SSID is defined in the menuconfig the issue doesn't happen.

The initial PR https://github.com/apache/nuttx/pull/7616 was the initial guilt, but later PR increased the issue.

With only the original PR if we try to ping from computer to the esp32 board the ping will not work, but I noticed that when we run at same time ping from board "nsh>" to some external IP, magically the ping from PC works monumentally and then start to DUP until read ~1010ms and succeed a single ping and start increasing the DUP timeout until reach 1010ms and a successful ping happen and the cycle restarts:

```
nsh> ping 8.8.8.8
PING 8.8.8.8 56 bytes of data
56 bytes from 8.8.8.8: icmp_seq=0 time=60.0 ms
56 bytes from 8.8.8.8: icmp_seq=1 time=40.0 ms
56 bytes from 8.8.8.8: icmp_seq=2 time=40.0 ms
56 bytes from 8.8.8.8: icmp_seq=3 time=40.0 ms
56 bytes from 8.8.8.8: icmp_seq=4 time=50.0 ms
56 bytes from 8.8.8.8: icmp_seq=5 time=40.0 ms
56 bytes from 8.8.8.8: icmp_seq=6 time=40.0 ms
56 bytes from 8.8.8.8: icmp_seq=7 time=40.0 ms
56 bytes from 8.8.8.8: icmp_seq=8 time=40.0 ms
56 bytes from 8.8.8.8: icmp_seq=9 time=40.0 ms
10 packets transmitted, 10 received, 0% packet loss, time 10100 ms
rtt min/avg/max/mdev = 40.000/43.000/60.000/6.403 ms
```

From PC to computer:
```
64 bytes from 192.168.0.7: icmp_seq=170 ttl=64 time=1000 ms (DUP!)
64 bytes from 192.168.0.7: icmp_seq=170 ttl=64 time=1000 ms (DUP!)
64 bytes from 192.168.0.7: icmp_seq=170 ttl=64 time=1003 ms (DUP!)
64 bytes from 192.168.0.7: icmp_seq=170 ttl=64 time=1003 ms (DUP!)
64 bytes from 192.168.0.7: icmp_seq=171 ttl=64 time=4.33 ms
...
64 bytes from 192.168.0.7: icmp_seq=171 ttl=64 time=135 ms (DUP!)
64 bytes from 192.168.0.7: icmp_seq=171 ttl=64 time=135 ms (DUP!)
64 bytes from 192.168.0.7: icmp_seq=171 ttl=64 time=136 ms (DUP!)
64 bytes from 192.168.0.7: icmp_seq=172 ttl=64 time=8.97 ms
...
64 bytes from 192.168.0.7: icmp_seq=172 ttl=64 time=1004 ms (DUP!)
64 bytes from 192.168.0.7: icmp_seq=172 ttl=64 time=1004 ms (DUP!)
64 bytes from 192.168.0.7: icmp_seq=172 ttl=64 time=1010 ms (DUP!)
64 bytes from 192.168.0.7: icmp_seq=172 ttl=64 time=1011 ms (DUP!)
64 bytes from 192.168.0.7: icmp_seq=173 ttl=64 time=12.3 ms
...
bytes from 192.168.0.7: icmp_seq=174 ttl=64 time=1007 ms (DUP!)
64 bytes from 192.168.0.7: icmp_seq=174 ttl=64 time=1012 ms (DUP!)
64 bytes from 192.168.0.7: icmp_seq=174 ttl=64 time=1012 ms (DUP!)
64 bytes from 192.168.0.7: icmp_seq=175 ttl=64 time=13.0 ms
```

After applying remaining PRs (master HEAD) the ping only works momentarily during the "nsh>" ping and after it finishes the PC ping always returns DUP timeout.

tmedicci, 2022-12-06:
> With only the original PR if we try to ping from computer to the esp32 board the ping will not work, but I noticed that when we run at same time ping from board "nsh>" to some external IP, magically [...]

Adding information about Alan's test: the commit that breaks the communication is https://github.com/apache/nuttx/commit/e002edf87c8e6765be59158e86fc8e651ae31a51

After it, pinging the device (ESP32) from an external device (PC) only returns if another ping from the device (to any target) is running.

anchao, 2022-12-06:
emm... sorry for the regression, It should be a driver compatibility issue, but I don’t have esp32 board on hand,  I will fix that when I get back to the office tomorrow.

anchao, 2022-12-07:
@acassis @tmedicci @lucasssvaz ,

Please try https://github.com/apache/nuttx/pull/7807, it will resolve esp32/wlan related issues

masayuki2009, 2022-12-08:
@anchao 
I noticed that similar performance regression happens with the virtnet (e.g. `qemu-armv8a:netnsh`)

```
After applying net/l2/l3/l4: add support of iob offload

+ /home/ishikawa/opensource/QEMU/qemu-7.1/build/qemu-system-aarch64 -cpu cortex-a53 -smp 4 -nographic -machine virt,virtualization=on,gic-version=3 -chardev stdio,id=con,mux=on -serial chardev:con -global virtio-mmio.force-legacy=false -netdev user,id=u1,hostfwd=tcp:127.0.0.1:10023-10.0.2.15:23,hostfwd=tcp:127.0.0.1:15001-10.0.2.15:5001 -device virtio-net-device,netdev=u1,bus=virtio-mmio-bus.0 -mon chardev=con,mode=readline -kernel ./nuttx
telnetd [4:100]

NuttShell (NSH) NuttX-10.4.0
nsh> uname -a
NuttX  10.4.0 34d2cde8a8 Dec  8 2022 10:40:53 arm64 qemu-armv8a
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK   STACK   USED  FILLED COMMAND
    0     0   0 FIFO     Kthread N-- Ready              00000000 008144 000976  11.9%  Idle Task
    1     1 192 RR       Kthread --- Waiting  Semaphore 00000000 008096 000960  11.8%  hpwork 0x402d3428
    2     2 100 RR       Kthread --- Waiting  Semaphore 00000000 008096 000960  11.8%  lpwork 0x402d3458
    3     3 100 RR       Task    --- Running            00000000 008112 002736  33.7%  nsh_main
    4     4 100 RR       Task    --- Waiting  Semaphore 00000010 008112 001616  19.9%  telnetd
nsh> iperf -c 43.31.77.50 -i 1 -t 10
       IP: 10.0.2.15

 mode=tcp-client sip=10.0.2.15:5001,dip=43.31.77.50:5001, interval=1, time=10

           Interval         Transfer         Bandwidth

   0.00-   1.01 sec    1769472 Bytes   14.02 Mbits/sec
   1.01-   2.02 sec    3604480 Bytes   14.53 Mbits/sec
   2.02-   3.03 sec    5455872 Bytes   14.66 Mbits/sec
   3.03-   4.04 sec    7208960 Bytes   13.89 Mbits/sec
   4.04-   5.05 sec    8929280 Bytes   13.63 Mbits/sec
   5.05-   6.06 sec   10731520 Bytes   14.28 Mbits/sec
   6.06-   7.07 sec   12582912 Bytes   14.66 Mbits/sec
   7.07-   8.08 sec   14450688 Bytes   14.79 Mbits/sec
   8.08-   9.09 sec   16302080 Bytes   14.66 Mbits/sec
   9.09-  10.10 sec   18137088 Bytes   14.53 Mbits/sec
   0.00-  10.10 sec   18137088 Bytes   14.37 Mbits/sec
iperf exit
```
```
Before applying net/l2/l3/l4: add support of iob offload

+ /home/ishikawa/opensource/QEMU/qemu-7.1/build/qemu-system-aarch64 -cpu cortex-a53 -smp 4 -nographic -machine virt,virtualization=on,gic-version=3 -chardev stdio,id=con,mux=on -serial chardev:con -global virtio-mmio.force-legacy=false -netdev user,id=u1,hostfwd=tcp:127.0.0.1:10023-10.0.2.15:23,hostfwd=tcp:127.0.0.1:15001-10.0.2.15:5001 -device virtio-net-device,netdev=u1,bus=virtio-mmio-bus.0 -mon chardev=con,mode=readline -kernel ./nuttx
telnetd [4:100]

NuttShell (NSH) NuttX-10.4.0
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK   STACK   USED  FILLED COMMAND
    0     0   0 FIFO     Kthread N-- Ready              00000000 008144 000976  11.9%  Idle Task
    1     1 192 RR       Kthread --- Waiting  Semaphore 00000000 008096 000960  11.8%  hpwork 0x402d3428
    2     2 100 RR       Kthread --- Waiting  Semaphore 00000000 008096 000960  11.8%  lpwork 0x402d3458
    3     3 100 RR       Task    --- Running            00000000 008112 002736  33.7%  nsh_main
    4     4 100 RR       Task    --- Waiting  Semaphore 00000010 008112 001616  19.9%  telnetd
nsh> iperf -c 43.31.77.50 -i 1 -t 10
       IP: 10.0.2.15

 mode=tcp-client sip=10.0.2.15:5001,dip=43.31.77.50:5001, interval=1, time=10

           Interval         Transfer         Bandwidth

   0.00-   1.01 sec   31866880 Bytes  252.41 Mbits/sec
   1.01-   2.02 sec   65421312 Bytes  265.78 Mbits/sec
   2.02-   3.03 sec   98975744 Bytes  265.78 Mbits/sec
   3.03-   4.04 sec  129941504 Bytes  245.27 Mbits/sec
   4.04-   5.05 sec  162316288 Bytes  256.43 Mbits/sec
   5.05-   6.06 sec  192970752 Bytes  242.81 Mbits/sec
   6.06-   7.07 sec  223592448 Bytes  242.55 Mbits/sec
   7.07-   8.08 sec  256802816 Bytes  263.05 Mbits/sec
   8.08-   9.09 sec  290242560 Bytes  264.87 Mbits/sec
   9.09-  10.10 sec  323715072 Bytes  265.13 Mbits/sec
   0.00-  10.10 sec  323715072 Bytes  256.41 Mbits/sec
iperf exit
```

anchao, 2022-12-08:
> @anchao I noticed that similar performance regression happens with the virtnet (e.g. `qemu-armv8a:netnsh`)

Got, I am busy on internal project today,  let me have a look and reply you later!