apache/nuttx #7812: SocketCAN broken since #7616

PetervdPerk-NXP, 2022-12-07:
Since the changes from PR #7616 it seems that SocketCAN support is borken.

When transmitting a can frame using cansend it will keep sending the message indefinitely, it seems that `priv->dev.d_len` stays != 0.

Reading a frame using candump returns `read: incomplete CAN frame`  it seems that `recvmsg()` returns 0 bytes.

xiaoxiang781216, 2022-12-07:
@anchao could you take a look?

PetervdPerk-NXP, 2022-12-07:
I've found the workaround to the transmit problem it seems with new IOB api we've to explicitely set d_len to 0. Not sure about the other drivers that are affected by this.

```patch
diff --git a/arch/arm/src/s32k3xx/s32k3xx_flexcan.c b/arch/arm/src/s32k3xx/s32k3xx_flexcan.c
index 3e17b52ee0..d0ac32894d 100644
--- a/arch/arm/src/s32k3xx/s32k3xx_flexcan.c
+++ b/arch/arm/src/s32k3xx/s32k3xx_flexcan.c
@@ -905,6 +905,8 @@ static int s32k3xx_transmit(struct s32k3xx_driver_s *priv)

   mb->cs = cs; /* Go. */

+  priv->dev.d_len = 0;
+
   regval = getreg32(priv->base + S32K3XX_CAN_IMASK1_OFFSET);
   regval |= mb_bit;
   putreg32(regval, priv->base + S32K3XX_CAN_IMASK1_OFFSET);
```

anchao, 2022-12-07:
> I've found the workaround to the transmit problem it seems with new IOB api we've to explicitely set d_len to 0. Not sure about the other drivers that are affected by this.

It seems to be the same issue as https://github.com/apache/nuttx/issues/7802 / https://github.com/apache/nuttx/pull/7807, I think d_len should be set to 0 after device callback,  I will submit a patch to resolve this issue from the common code

anchao, 2022-12-07:
@PetervdPerk-NXP  I resubmit the https://github.com/apache/nuttx/pull/7807, please try this change if it can resolves your issue

acassis, 2022-12-08:
@PetervdPerk-NXP could you please confirm it is working now and please close this issue

xiaoxiang781216, 2022-12-08:
@acassis @PetervdPerk-NXP reply in #7807. It work, so let's close it.

closed 2022-12-08 by xiaoxiang781216