apache/nuttx #8061: ESP32-C3 Wi-Fi miss-behavior depending on the Access Point

tmedicci, 2023-01-09:
**The Behavior**

After the implementation of the IOB buffer handling directly from the Wi-Fi driver (https://github.com/apache/nuttx/pull/7817/), it seems that there exist problems while handling small packets and ping requests do not return successfully.

It's worth noting, however, that this problem is depending on the connected access point. Android Hotspots and TP-Link (OpenWrt-enabled) routers usually fail. The Sagemcom F@ST 3895 router ping response is higher and, then, it doesn't fail.

**Steps to reproduce**

It's possible to see the behavior on master [67ccee2c4f581c3c772592f3393e437cf11af186](https://github.com/apache/nuttx/commit/67ccee2c4f581c3c772592f3393e437cf11af186) since the commit [61563d6004fe92ae50d567bc4f81c81f60d8c757](https://github.com/apache/nuttx/commit/61563d6004fe92ae50d567bc4f81c81f60d8c757)

`
make -j distclean && ./tools/configure.sh esp32c3-devkit:wapi && make menuconfig && make -j bootloader && make -j flash ESPTOOL_PORT=/dev/ttyUSB0 ESPTOOL_BINDIR=./ && minicom -D /dev/ttyUSB0
`

In order to check the possible problem regarding small packets + IOB implementation, just apply the following patch and turn on the debug logs for the network:

```
diff --git a/arch/risc-v/src/esp32c3/esp32c3_wlan.c b/arch/risc-v/src/esp32c3/esp32c3_wlan.c
index a2f2af4f89..8c678d2157 100644
--- a/arch/risc-v/src/esp32c3/esp32c3_wlan.c
+++ b/arch/risc-v/src/esp32c3/esp32c3_wlan.c
@@ -366,6 +366,8 @@ static int wlan_rx_done(struct wlan_priv_s *priv, void *buffer,
   irqstate_t flags;
   int ret = 0;
 
+  nwarn("Wlan receive packet length: %d\n", len);
+
   if (!priv->ifup)
     {
       goto out;
@@ -476,6 +478,7 @@ static void wlan_rxpoll(void *arg)
       dev->d_iob = iob;
       dev->d_len = iob->io_pktlen;
 
+      nwarn("d_len: %d", dev->d_len);
 #ifdef CONFIG_NET_PKT
 
       /* When packet sockets are enabled,
```

**Logs**
```
nsh> wapi psk wlan0 mypasswd 3
nsh> wapi essid wlan0 myssid 1
nsh> renew wlan0
wlan_rx_done: Wlan receive packet length: 342
wlan_rxpoll: d_len: 328
wlan_rx_done: Wlan receive packet length: 342
wlan_rxpoll: d_len: 328
nsh> ping 8.8.8.8
PING 8.8.8.8 56 bytes of data
wlan_rx_done: Wlan receive packet length: 42
wlan_rxpoll: d_len: 28
arp_in: ERROR: Packet Too small
No response from 8.8.8.8: icmp_seq=0 time=1000 ms
wlan_rx_done: Wlan receive packet length: 42
wlan_rxpoll: d_len: 28
arp_in: ERROR: Packet Too small
No response from 8.8.8.8: icmp_seq=1 time=1000 ms
wlan_rx_done: Wlan receive packet length: 42
wlan_rxpoll: d_len: 28
arp_in: ERROR: Packet Too small
wlan_rx_done: Wlan receive packet length: 42
wlan_rxpoll: d_len: 28
arp_in: ERROR: Packet Too small
No response from 8.8.8.8: icmp_seq=2 time=1000 ms
wlan_rx_done: Wlan receive pack
```

tmedicci, 2023-01-09:
@anchao , could you please take a look at it? Let me know if you need anything else.

anchao, 2023-01-10:
Hi @tmedicci ,

Thanks for your analysis, d_len should keep the l2(ethernet) header, I send a pull request to solve this issue, please review https://github.com/apache/nuttx/pull/8064

closed 2023-01-10 by gustavonihei