apache/nuttx #8098: FOC get Back EMF

shizacat, 2023-01-12:
Hi!
More the questionstion than the issue

I want to make motor control though FOC with get angle sensorless from BEMF.
For base I took example from 'example/foc'.
How I understood, right approach - get data about BEMF from injected ADC channels?
And now problems:

- The driver FOC havn't have this ability? For this needs to be done to add variable as for 'current' (foc_state_s, stm32_foc_data_s) and methods. That is, you need to update exactly NuttX?
- Set up a board to work in dual mode ADC, to get 4+4 injectd channels

Am I thinking in the right direction?

acassis, 2023-01-12:
@raiden00pl could you please help here?

zouboan, 2023-01-13:
@shizacat Neither sensored nor sensorless mode of NuttX FOC driver uses BEMF, so the FOC driver does not support the acquisition of motor phase voltage at present

zouboan, 2023-01-13:
@shizacat Yes, you should add voltage define in `struct stm32_foc_data_s` of arch/arm/stm32/stm32_foc.c
and you should implement `stm32_foc_volt_get` follows the design of `stm32_foc_curr_get`
and you should implement `board_foc_voltage_get` In b-g431b-esc1 follows the design of `board_foc_current_get`

raiden00pl, 2023-01-13:
I don't think you can directly sense BEMF in FOC.
BEMF sensing requires only two phases of the tree phases motor conducting at any point of time,
but in FOC you have three phases conducting.

Detecting BEMF is possible only for six-step commutation (also known as trapezoidal commutation) which is not currently supported with NuttX.

BEMF angle estimation methods for FOC use BEMF estimation techniques, where you can use phases
currents and motor model to estimate BEMF and further phase angle.
One of this method is implemented here (SMO observer): 
https://github.com/apache/nuttx-apps/blob/master/industry/foc/float/foc_ang_osmo.c
https://github.com/apache/nuttx/blob/1f95cfdde8338f99116f64d59db43ca146d791ae/libs/libdsp/lib_observer.c

zouboan, 2023-01-13:
@raiden00pl maybe he want to sense BEMF when three-phase bridge was shut off, then we can sense BEMF and observer angle by fluxlinkage observer in case of motor power off.
This method is generally used in case when the motor was driven by external torque when motor power off, if we can tracking electric angle under this working condition, it will be easier to do a sensorless startup at the next time.

shizacat, 2023-01-14:
Ohh, how many ))
Thank you, for your time!

@acassis, Yes/no. I rewrite this example for me. I found it complicated and not very clear. But all the changes only control (FMS), main idea of the code unchanged.

@zouboan Yep, I want change it ... try.
@zouboan Thank you for tips.

@raiden00pl Why not, Back EMF it simply voltage on ADC, which the FOC driver picks up by analogy with the current. Thank you for tips.

@raiden00pl Yes, It will be great. This is what I want.

raiden00pl ... FOC driver pick up physical state of motor as current or else, and BEMF not very strong different the current. I think, it right place to take the current and BEMF in one place.

But the calculation of the phase angle itself is already located somewhere from the outside...

Why not create separate driver for BEMF sensing with its own ADC and triggering ? - Timing, Current and voltage (BEMF) must be taken at the same time - this is the state of the motor.
Is there any coupling between the FOC driver and BEMF sensing that can cause problems here? - Yep, timing.

raiden00pl, 2023-01-15:
> raiden00pl ... FOC driver pick up physical state of motor as current or else, and BEMF not very strong different the current. I think, it right place to take the current and BEMF in one place.

That's not entirely true. In fact, the FOC driver knows nothing about the powered motor. What the driver know is the state of 3-phase bridge that powers the motor (which is PWM duty and current on shunts).
That's why the driver is completely separated from parameters like the motor phase angle, motor temperature or VBUS.
This separation is very important for modular architecture, that is why I was slightly skeptical about measuring BEMF in the FOC driver. But after some thinking, I'm OK with this as long as everything will be hidden behind the Kconfig option.
I'll look at this topic more in the week, especially how to implement BEMF sensing for STM32.

closed 2023-01-20 by pkarashchenko

raiden00pl, 2023-01-20:
All uper-half logic is in place now but lower-level logic for stm32 is not yet upstream. I have some initial code but not tested  on HW yet.

acassis, 2023-01-22:
Nice work @raiden00pl !!! Kudos!!!

shizacat, 2023-01-22:
@raiden00pl  NIce work!