apache/nuttx #8147: Add support for an LTE driver on NuttX

duduita, 2023-01-16:
I need to add support for a Quectel BG770A modem.

After looking at this [commit](https://github.com/apache/nuttx/commit/dc5d8f7c4478efee10c661034600a61d52d2c13f), I realized that NuttX has native code that handles some kind of modems, not requiring the creation of a driver from scratch. But, maybe it does not cover all types of modems, since NuttX has a separate driver for the [ALT1250 modem](https://github.com/apache/nuttx/blob/master/drivers/modem/alt1250/alt1250.c).

Then, I’m not sure if I’ll need to create a driver from a scratch, as the driver for the ALT1250 modem, or if I can just configure it, as it was done in this [commit](https://github.com/apache/nuttx/commit/dc5d8f7c4478efee10c661034600a61d52d2c13f).

Could anyone help me to decide which approach I’ll have to follow?

jerpelea, 2023-01-17:
@duduita 
unfortunately is not a straight answer but you may want to consider the following aspects
- If your modem HW uses standard communication without extra functionality you can use the existing driver and configure it for your modem. 
- If your modem HW implements extra functionality you may want to create a new driver that will expose that extra functionality.

duduita, 2023-01-17:
Could give me an example that what would be an extra functionality?

jerpelea, 2023-01-17:
Unfortunately those extra features are dependent on each modem, usually modems have special commands for special features

duduita, 2023-01-26:
Thank you for your explanation @jerpelea!

closed 2023-01-26 by duduita

duduita, 2023-03-14:
Just to let you know, that I could get connected using an LTE-M module (Quectel BG770A-GL) with NuttX, by using the PPP protocol.

Basically, I had to use the PPPD app on NuttX, and change a few things on the `connect_script` on the `pppd_main.c`

```
static FAR const char connect_script[] =
  "ECHO ON "
  "TIMEOUT 30 "
  "\"\" AT+CGDCONT=1,\\\"IP\\\",\\\"virtueyes.com.br\\\" "
  "OK AT+COPS=1,2,\\\"72410\\\",7 "
  "OK ATD*99# "
  "CONNECT \\r\\c";

static FAR const char disconnect_script[] =
  "\"\" ATZ "
  "ERROR \\r\\c";
```

I only had to define my PDP context, use AT+COPS to get registered and connected, and use ATD*99# to start the dial-up on the modem.

To define the PDP context (replace the variables according to your internet provider):
`AT+CGDCONT=1,“IP”,“virtueyes.com.br”`
And use the `CONFIG_NETUTILS_PPPD_PAP`, setting the pap authentication in the `pppd_main.c`:

```
#ifdef CONFIG_NETUTILS_PPPD_PAP
    .pap_username = "virtu",
    .pap_password = "virtu",
#endif 
```

To get registered manually:
`AT+COPS=1,2,“72410”,7
`
A hint to see your internet provider ID is using the AT+COPS=?, since it will show all the networks your module sees.

If you are not seeing anything, you can set the band used in your country, for instance, in Brazil we use Band 3 (0x4) for CATM1 and Band 28 (0x8000000) for NB-IOT, so you need to find the correspondent hex numbers in the datasheet to use in the following command:
`AT+QCFG=“band”,0x0,0x4,0x8000000
`
Finally, the dial-up command:
`ATD*99#
`

Another important thing: some modems as BG770A-GL requires the `\\r` at the end of the `connect_script` (Carriage Return).

I was wondering if it would be possible to use LTE-M modems with PPP on NuttX, but I haven't found any tutorials involving that, it's why I decided to explain in detail what I've done to do that.