apache/nuttx #8207: usleep accuracy

maxikrie, 2023-01-20:
I am using a simple usleep(100); and I noticed that the actual delay is 140 us. My specific configuration is a tickless scheduler where USEC_PER_TICK is 1. 

For comparison I have set up a thread that waits on sigwaitinfo(&mask, &info);, which is signaled by a hardware timer with a timeout of 100 us. Using clock_systime_ticks() for timing I see that this approach almost gets precisely done what I am expecting.

From what I am understanding usleep employs signals as well under the hood, however using clock_systime_ticks()  for analysis it seems that it performs significantly worse. Is this expected due to the overhead?

maxikrie, 2023-01-21:
This minimal example illustrates the problem:

int main(int argc, FAR char *argv[])
{
  printf("Hello, World!!\n");

  uint32_t N = 1000;
  uint32_t time[N];
  useconds_t sleep_time = 100;

  struct timespec t_start;
  clock_gettime(0, &t_start);
  for(int i=0; i < N; i++)
  {
	  struct timespec t1;
	  clock_gettime(0, &t1);
	  usleep(sleep_time);
	  struct timespec t2;
	  clock_gettime(0, &t2);
	  struct timespec dt;
	  clock_timespec_subtract(&t2, &t1, &dt);
	  time[i] = dt.tv_nsec/1000;
  }
  struct timespec t_end;
  clock_gettime(0, &t_end);

  struct timespec dt;
  clock_timespec_subtract(&t_end, &t_start, &dt);
  printf("Average sleep time: %ld (%ld)\n",  dt.tv_nsec/1000/N, sleep_time);

  for(int i=0; i < N; i++)
  {
	  printf("Iteration %d: sleep time %ld (%ld)\n", i, time[i], sleep_time);
  }
  return 0;
}

On the stm32f4discovery:usbnsh (with CONFIG_SCHED_TICKLESS=y and CONFIG_USEC_PER_TICK=1) I am getting 124 us sleep instead of 100 us, which I find uprising.

maxikrie, 2023-01-21:
This however results in the desired behavior

```
	printf("Hello, World!!\n");

	uint32_t N = 1000;
	uint32_t time[N];
	useconds_t sleep_time = 100;

	int tmr_id = open("/dev/timer0", O_RDONLY);
	if (tmr_id < 0)
	{
		fprintf(stderr, "ERROR: Failed to open timer: %d\n", errno);
		close(tmr_id);
		return EXIT_FAILURE;
	}

	ioctl(tmr_id, TCIOC_SETTIMEOUT, 100);

	struct timer_notify_s notify;

	notify.pid = getpid();
	notify.periodic = true;

	notify.event.sigev_notify = SIGEV_SIGNAL;
	notify.event.sigev_signo = SIGUSR1;
	notify.event.sigev_value.sival_ptr = NULL;

	int ret = ioctl(tmr_id, TCIOC_NOTIFICATION,
			(unsigned long) ((uintptr_t) &notify));
	if (ret < 0)
	{
		fprintf(stderr, "ERROR: Failed to set the timer handler: %d\n", errno);
		close(tmr_id);
		return EXIT_FAILURE;
	}

	ret = ioctl(tmr_id, TCIOC_START, 0);
	if (ret < 0)
	{
		fprintf(stderr, "ERROR: Failed to start the timer: %d\n", errno);
		close(tmr_id);
		return EXIT_FAILURE;
	}

	sigset_t mask;
	sigaddset(&mask, SIGUSR1);

	struct timespec t1;
	clock_gettime(0, &t1);

	struct timespec t_start;
	clock_gettime(0, &t_start);
	for (int i = 0; i < N; i++)
	{
		siginfo_t info;
		int bla = sigwaitinfo(&mask, &info);
		struct timespec t2;
		clock_gettime(0, &t2);
		struct timespec dt;
		clock_timespec_subtract(&t2, &t1, &dt);
		t1 = t2;
		time[i] = dt.tv_nsec / 1000;
	}

	struct timespec t_end;
	clock_gettime(0, &t_end);

	struct timespec dt;
	clock_timespec_subtract(&t_end, &t_start, &dt);
	printf("Average sleep time: %ld (%ld)\n", dt.tv_nsec / 1000 / N,
			sleep_time);

	for (int i = 0; i < N; i++)
	{
		printf("Iteration %d: sleep time %ld (%ld)\n", i, time[i], sleep_time);
	}
	return 0;
```

maxikrie, 2023-01-23:
I have investigated this a bit further and both sigwaitinfo and usleep employ sig_timedwait. However calling sig_timedwait from usleep requires setting up a watchdog with the specified usleep time, which causes additional delay on the order of several microseconds (if interested the if statement in sig_timedwait:315 is evaluated differently when called from usleep compared to sigwaitinfo). 

Shouldn't there be a better way along the lines of getting the current time in ticks when calling usleep and setting a time in the future depending on the usleep value to wake up the task instead of setting a watchdog with the usleep value causing unwanted additional delay?

maxikrie, 2023-01-23:
I can of course simply employ sigwaitinfo as described above, however this requires an additional hardware timer, which dependent on the hardware and use case may not be available.

xiaoxiang781216, 2023-02-11:
@maxikrie could you create a PR? It's definitely to fix the root cause, instead wasting a hardware timer.