apache/nuttx #8325: RNDIS/CDC-ECM/CDC-NCM

TimJTi, 2023-01-28:
I have been trying to get RNDIS working but cannot get it to connect to me Win11 machine. It won't connect to my Ubuntu 24.04 machine either. Creating this as an issue as per suggestion by @masayuki2009 

CDC-ECM works with Ubuntu straight away but there are no built-in drivers for this with Windows so it is not an option right now.

Further investigation shows that RNDIS is likely to be deprecated from Linux in the near future anyway; @hartmannathan provided this link: https://www.phoronix.com/news/Linux-Disabling-RNDIS-Drivers

One solution might be to implement CDC-NCM (suggested by @michi-jung as that is then universal to Windows, Linux and Mac I believe?

Otherwise it looks like RNDIS is broken - unless anyone knows differently using recent NuttX builds, Windows and/or Linux?

masayuki2009, 2023-01-28:
spresense:rndis and rndis_smp have been working fine with ubuntu 18.04 x86_64 on VMware Fusion.

TimJTi, 2023-01-28:
> spresense:rndis and rndis_smp have been working fine with ubuntu 18.04 x86_64 on VMware Fusion.

With the defconfigs in the boards folders of the NuttX distribution or your own? I have been comparing my .config with the few in the distribution that have CONFIG_RNDIS=y and see no obvious differences. I tried RNDIS a year ago with Ubuntu 20.,04 and it didn't work then - so maybe there's a change since 18.04 and in later Windows builds that has broken NuttX RNDIS?

Initial investigations actually suggest an issue with arp requests failing - requests from my board don't seem to get out, and Windows doesn't seem to be sending any (but I'm a Wireshark newbie so may not be interpreting it right). I'm yet to try enabling DNS so perhaps that might be the key difference.

acassis, 2023-01-28:
I posted in the mailing list how to get it working, I will duplicate the steps here for future reference:

```

Configure your board:
=====================

$ ./tools/configure.sh stm32f4discovery:rndis

Compile NuttX to your board:
============================

$ make -j

Flash to your board:
====================

$ sudo openocd -f interface/stlink.cfg -f target/stm32f4x.cfg -c init -c "reset halt" -c "flash write_image erase nuttx.bin 0x08000000"
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 2000 kHz
Info : STLINK V2J14S0 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.203144
Info : stm32f4x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f4x.cpu on 3333
Info : Listening on port 3333 for gdb connections
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08000188 msp: 0x20003f24
Info : device id = 0x10036413
Info : flash size = 1024 kbytes
auto erase enabled
wrote 262144 bytes from file nuttx.bin in 11.043253s (23.182 KiB/s)

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections

Reset your board
================

Run dmesg to confirm that USB RNDIS device was detected:
========================================================

[ 1099.821480] usb 3-3: new full-speed USB device number 12 using xhci_hcd
[ 1099.972379] usb 3-3: New USB device found, idVendor=584e, idProduct=5342, bcdDevice= 0.01
[ 1099.972389] usb 3-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1099.972393] usb 3-3: Product: RNDIS gadget
[ 1099.972396] usb 3-3: Manufacturer: NuttX
[ 1099.972398] usb 3-3: SerialNumber: 1234
[ 1099.988952] usbcore: registered new interface driver cdc_ether
[ 1099.990144] rndis_host 3-3:1.0: skipping garbage
[ 1099.990641] rndis_host 3-3:1.0: dev can't take 1558 byte packets (max 660), adjusting MTU to 602
[ 1099.992089] rndis_host 3-3:1.0 eth0: register 'rndis_host' at usb-0000:00:14.0-3, RNDIS device, a0:e0:de:ad:ca:fe
[ 1099.992102] usbcore: registered new interface driver rndis_host
[ 1099.994026] usbcore: registered new interface driver rndis_wlan
[ 1099.997001] rndis_host 3-3:1.0 enxa0e0deadcafe: renamed from eth0

Configure your Ubuntu or other Linux distro to share network:
=============================================================

Click in the top right corner of your Ubuntu and go to:

NuttX Ethernet -> Wired Settings

Click in the 'Gear icon' and in the tab "IPv4" select: "Shared to other computers"

Click on "Apply"

Disconect and connect the USB cable to force it to get IP.

Verify which IP your board got:
===============================

$ tail -f /var/log/syslog
...
Jan 28 10:30:24 dev dnsmasq-dhcp[35526]: DHCPDISCOVER(enxa0e0deadcafe) 00:e0:de:ad:ca:fe 
Jan 28 10:30:24 dev dnsmasq-dhcp[35526]: DHCPOFFER(enxa0e0deadcafe) 10.42.0.86 00:e0:de:ad:ca:fe 
Jan 28 10:30:24 dev dnsmasq-dhcp[35526]: DHCPREQUEST(enxa0e0deadcafe) 10.42.0.86 00:e0:de:ad:ca:fe 
Jan 28 10:30:24 dev dnsmasq-dhcp[35526]: DHCPACK(enxa0e0deadcafe) 10.42.0.86 00:e0:de:ad:ca:fe nuttx
Jan 28 10:30:29 dev systemd[1]: NetworkManager-dispatcher.service: Deactivated successfully.
^C

Ping this IP to confirm it is working:
======================================

alan@dev:~/nuttxspace/nuttx$ ping 10.42.0.86
PING 10.42.0.86 (10.42.0.86) 56(84) bytes of data.
64 bytes from 10.42.0.86: icmp_seq=1 ttl=64 time=0.809 ms
64 bytes from 10.42.0.86: icmp_seq=2 ttl=64 time=0.849 ms
^C
--- 10.42.0.86 ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 1027ms
rtt min/avg/max/mdev = 0.809/0.829/0.849/0.020 ms

Connect to your board over telnet:
==================================

alan@dev:~/nuttxspace/nuttx$ telnet 10.42.0.86
Trying 10.42.0.86...
Connected to 10.42.0.86.
Escape character is '^]'.

NuttShell (NSH) NuttX-12.0.0
nsh> 
```

TimJTi, 2023-01-28:
Please do not close. This is not resolved for me, yet.

TimJTi, 2023-01-28:
Thank you for reopening. I will continue trying to get this working next week Assuming I do, hopefully whatever the reason and fix is will prove useful to others.

masayuki2009, 2023-01-28:
@TimJTi 

>With the defconfigs in the boards folders of the NuttX distribution or your own? I

I've been using the latest upstream.

>I tried RNDIS a year ago with Ubuntu 20.,04 and it didn't work then - so maybe there's a change since 18.04 and in later Windows builds that has broken NuttX RNDIS?

I've just installed ubuntu 22.04 x86_64 and confirmed that RNDIS works with spresense
So I think ubuntu 20.04 should work.

TimJTi, 2023-01-28:
@masayuki2009 

Thank you for taking the trouble to try this - I am using 22.04 so it has to be down to my .config. I have more things to try, now, and will report back what I did to fix it so hopefully others will benefit from my struggles!

acassis, 2023-01-28:
I forgot to comment, but I'm also using Ubuntu 22.04.1 LTS

TimJTi, 2023-01-28:
> I forgot to comment, but I'm also using Ubuntu 22.04.1 LTS

Thanks - means, even more so, that it's a question of my .config :( My hope is that it's down to not using DNS.

TimJTi, 2023-01-30:
Initial results today, on Ubuntu which, I now remember, is exactly what I saw > 1 year ago and I gave up at that time as it wasn't important then. It is now :(

[282860.884239] usb 1-2: USB disconnect, device number 28
[282861.360379] usb 1-2: new high-speed USB device number 29 using xhci_hcd
[282861.509252] usb 1-2: New USB device found, idVendor=584e, idProduct=5342, bcdDevice= 0.01
[282861.509258] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[282861.509261] usb 1-2: Product: RNDIS gadget
[282861.509263] usb 1-2: Manufacturer: NuttX
[282861.509265] usb 1-2: SerialNumber: 1234
[282861.510911] rndis_host 1-2:1.0: skipping garbage
[282861.511154] rndis_host 1-2:1.0: dev can't take 1558 byte packets (max 1542), adjusting MTU to 1484
[282861.511483] rndis_host 1-2:1.0: RNDIS_MSG_QUERY(0x01010101) failed, -32
[282861.511489] rndis_host 1-2:1.0: rndis get ethaddr, -32
[282861.511584] rndis_host: probe of 1-2:1.0 failed with error -32
[282861.511830] rndis_wlan 1-2:1.0: skipping garbage
[282861.511989] rndis_wlan 1-2:1.0: dev can't take 1558 byte packets (max 1542), adjusting MTU to 1484

A web search for that RNDIS_MSG_QUERY fail shows it has been seen by others on different boards and in different situations but still looking...but my belief is it is due to a combination of latest Ubuntu and Windows RNDIS drivers and a possible shortcoming of the SAMA5-specific usb device driver which is why I see it but stm32 users (for example) don't.

masayuki2009, 2023-01-31:
@TimJTi 

> A web search for that RNDIS_MSG_QUERY fail shows it has been seen by others on different boards and in different situations but still looking...but my belief is it is due to a combination of latest [...]

I think that there's something wrong with the SAMA5-specific USB driver.
Actually, when I tried to make USB RNDIS work with spresense, I had to fix the driver even it worked for Mass Storage.

You can find the fix as follows.

```
commit 06aa50e9bb255284a3fc286d466df8411a562148
Author: Masayuki Ishikawa <masayuki.ishikawa@gmail.com>
Date:   Thu Jun 13 05:40:03 2019 +0000

    Merged in masayuki2009/nuttx.nuttx/rndis_on_spresense (pull request #895)

    rndis on spresense

    * arch/arm/src/cxd56xx: Fix ep0 out packet handling in cxd56_usbdev.c

        This change fixes to handle ep0 out packet which exceeds EP0_MAXSIZE.

        For example, 76 bytes of out packet is sent from rndis host during
        rndis negotiation and finally the packet is split into two out packets
        (64 bytes out packet + 12 bytes out packet). To handle these packets,
        actual ep0reqlen is stored during ep0 out setup stage, and if all
        out packets are received, it dispatches to cxd56_ep0setup(), otherwise,
        it sets up the internal USB DMA to receive remaining out packets.

        Signed-off-by: Masayuki Ishikawa <Masayuki.Ishikawa@jp.sony.com>

    * configs/spresense: Add support for rndis configuration

        Signed-off-by: Masayuki Ishikawa <Masayuki.Ishikawa@jp.sony.com>

    Approved-by: Gregory Nutt <gnutt@nuttx.org>
```

TimJTi, 2023-01-31:
@masayuki2009 Perfect! That's exactly what I believe to be going on, based on lots of evidence on other processors an RTOS'

I will try and implement similar fix and create a PR.

closed 2023-02-09 by xiaoxiang781216