apache/nuttx #8327: a little question of esp32c3

flyingfish89, 2023-01-28:
I found a problem：
  at this function: nuttx\sched\clock\clock_initialize.c\clock_initialize
  if we use ./tools/configure.sh -l  esp32c3-devkit:nsh and ./tools/configure.sh -l esp32c3-devkit:timer  and other frameworks
  i found that nuttx\arch\risc-v\src\esp32c3\esp32c3_timerisr.c\up_timer_initialize, even though it assigned something to the clock 
  interrupt,  but the clock interrupt didn't run
  
  I looked at the interrupt list and found that the clock interrupt existed but was not called.
  like this :
 
[image]

the irq = 37 is an interruption of stdin,stdout,the clock interrupt is irq = 53(at the top of the text clock init is successful and it didn't run)
@acassis Does the clock interrupt need to turn on any configuration

acassis, 2023-01-28:
@flyingfish89 was this timer IRQ working before my modification?

flyingfish89, 2023-01-28:
> @flyingfish89 was this timer IRQ working before my modification?
I found it last week

acassis, 2023-02-02:
@flyingfish89 please test https://github.com/apache/nuttx/pull/8415 and let me know if it fixed the issue

closed 2023-02-03 by flyingfish89