apache/nuttx #8491: USB HID Keyboard causes crash

TimJTi, 2023-02-09:
SAMA5D2 processor.

With the USB HID Keyboard driver enabled, a sequence of 2 or 3 attach-detach cycles causes a crash. Other USB host drivers (e.g. Mass Storage) behave OK.

This is the first stack dump crash I've encountered, and happy to investigate further with a bit of guidance, or maybe someone knows right away what the issue is? I have half a suspicion that it is the OHCI Monitor based on a message I saw once.

```
up_dump_register: R0: 6000005f R1: 00400416 R2: 00000000  R3: 20058c00
up_dump_register: R4: 00000000 R5: 00000000 R6: 00000008  R7: 00000000
up_dump_register: R8: 2005462c SB: 00000008 SL: ffffff80  FP: 00000000
up_dump_register: IP: 00000000 SP: 20064028 LR: 2001e5e0  PC: 2001e624
up_dump_register: CPSR: 200000df
show_tasks:    PID   PRI    STACKBASE STACKSIZE   COMMAND
show_tasks:   ----   ---   0x20055000      1024   irq
dump_task:      0      0   0x2005ca40      1000   Idle Task
dump_task:      1    192   0x2005e2b8      1992   hpwork 0x200573dc
dump_task:      2    192   0x2005eec8      1992   hpwork 0x200573dc
dump_task:      3    100   0x2005fad8       968   lpwork 0x20057400
dump_task:      4    100   0x200602e8       968   lpwork 0x20057400
dump_task:      5    100   0x20060b08       968   lpwork 0x20057400
dump_task:      6    100   0x20061318       968   lpwork 0x20057400
dump_task:      8    110   0x20062d30      2000   FUSB302 Monitor
dump_task:      9    100   0x20063960      2000   OHCI Monitor
dump_task:     10    100   0x20064570      2000   EHCI Monitor
dump_task:     11    100   0x200722a8      4056   nsh
```

acassis, 2023-02-14:
@TimJTi Did you test it on STM32F4Discovery board to confirm the issue is not in your arch driver?

TimJTi, 2023-02-14:
@acassis - I only have sama5 boards and not sure if my eval boards are still usable! My thinking was that since the USB memory stick works 100% reliably and I can plug/unplug many times without issue it was unlikely to be the SAMA5 usb code - but it is possible. I think one is ohci and the other ehci so i will investigate further. After all, I did find an issue with the SAMA5 device driver so further issues are I guess rather likely.

I did trace the connect, enumeration, and detach events and they superficially seemed to make sense. the keyboard actually works the first time, and it "feels" perhaps like a big chunk of memory is not being released.

What I've not got a good handle on is debugging full blown crashes as the screenful's of register dumps seem rather pointless!

ldube, 2023-03-23:
@TimJTi  Check and see if #8879 fixes this problem.

ldube, 2024-01-12:
@TimJTi Please close this issue.

TimJTi, 2024-01-12:
@ldube
With apologies, I haven't been able to verify it yet for various reasons. I can close if wanted but if I may end up reopening if it's still an issue.

What would your prefer - wait, pending me checking in the next month or so, or close it with the potential of it being re-opened?

ldube, 2024-01-12:
> What would your prefer - wait, pending me checking in the next month or so, or close it with the potential of it being re-opened?

Wait.

casaroli, 2026-11-08:
Possibly fixed by #8879.

acassis, 2026-11-08:
@TimJTi did you have time to check again?

TimJTi, 2026-11-08:
> [@TimJTi](https://github.com/TimJTi) did you have time to check again?

@acassis - my wife died earlier this year having being diagnosed with terminal cancer last December. I an no longer actively developing products and I'm afraid I never did get to check this.

acassis, 2026-11-08:
> > [@TimJTi](https://github.com/TimJTi) did you have time to check again?
> 
> [@acassis](https://github.com/acassis) - my wife died earlier this year having being diagnosed with terminal cancer last December. I an no longer actively developing products and I'm afraid I never did get to check this.

Hi Tim,
I'm very sad to hear that! Sometimes I feel guilt because I could spend more time together with my family and I'm here helping this project (that only a few people care about). But for some strange reason I think it makes sense and I think I'm doing the right thing. And I really enjoy doing it. So, I think instead of spend much time with them I will focus on having quality time with them!

TimJTi, 2026-11-08:
> Hi Tim, I'm very sad to hear that! Sometimes I feel guilt because I could spend more time together with my family and I'm here helping this project (that only a few people care about). But for some [...]

Thank you @acassis . My wife and I ran the business together and I have lost my Mojo for its products, understandably I think. It is VERY important to spend time with your family - as much as possible - as you can never know. MAKE SURE YOU DOI!!!

I have very much enjoyed all my interactions with the NuttX community and it is a superb OS. I always cared about it and still do - I have not unsubscribed from mailing lists for example. I am starting a few personal projects and may very will be using NuttX again (but not on SAMA5D2).

Shall I close this or would you like to keep it open?

Take care.

acassis, 2026-11-08:
Thank you @TimJTi yes I think you can close it. I used USB HID keyboard recently and everything is working fine now. Thank you very much for your kind words!

closed 2026-11-08 by TimJTi