apache/nuttx #8532: MSG_PEEK support

fjpanag, 2023-02-14:
As far as I can tell `MSG_PEEK` is not supported by TCP sockets.

It would be great to add support for this feature.

This flag can be very useful when the size of the incoming packet is unknown, and it is encoded in the packet itself.  
You can "peek" into the data, determine the size of the packer, allocate the needed space, and then perform the actual read.

A great example of this is MQTT.  
In MQTT you have a header byte, 1 to 4 bytes that encode the size of the payload, and then the payload.  

Now, these bytes have to be read one-by-one, allocate the space and then read the payload.  
This is not optimal, because if the packet has not been received yet completely (or malloc fails), it is difficult to try again later.  
(You have to keep somewhere the already read data to reuse later - complicated).

With `MSG_PEEK` you can read bytes 1-4 to determine the size, check that everything is correct, and then read in one go.  
If everything goes wrong, just try again later. Everything is still in the buffer, untouched.

xiaoxiang781216, 2023-02-15:
Look it's useful.

fjpanag, 2023-08-04:
This was addressed in #9791.

closed 2023-08-04 by fjpanag