apache/nuttx #8537: [RFC] Nim language support

centurysys, 2023-02-15:
I started porting Nim language to NuttX,  a simple web server is up and running.

https://github.com/centurysys/nuttx-apps/blob/MAS1xx/master/examples/hello_nim/webserver.nim

```nim
import std/asyncdispatch
import std/strformat
import jester

router myrouter:
  get "/hello/@name":
    let username = @"name"
    resp &"Hello, {username} !"

proc runserver(): Future[void] {.async.} =
  let port = 80.Port
  let settings = newSettings(port = port)
  var server = initJester(myrouter, settings = settings)
  server.serve()

proc run_http_server() {.exportc, cdecl.} =
  waitFor runserver()
```
(I have also confirmed the operation of other libraries, such as the zstandard library.)

I need to configure Makefile and Nim's fine settings (malloc allocation size, cpu kind, optimization flags, etc.) in a file for Nim. I am wondering how to incorporate this into the core like Zig/Rust.

https://github.com/centurysys/nuttx-apps/tree/MAS1xx/master/examples/hello_nim

(The file config.nims in the above directory is the file that needs to be set.)

I don't think the Nim language changes are complete, but submitting a pull request to the community.

https://github.com/nim-lang/Nim/pull/21372

I would be grateful for your advice.

xiaoxiang781216, 2023-02-15:
maybe you can follow how Zig/Rust integrate into NuttX.

centurysys, 2023-02-15:
@xiaoxiang781216 Thank you for reply!

I was thinking that I would have to do some work on Toolchain.defs, etc like Zig/Rust does.
I remembered that Nim has a VM called nimscript that can be run at compile time(config.nims). I will investigate how to parse .config with nimscript and change the configuration.

centurysys, 2023-02-19:
I'm considering the following file to extract settings from **.config** and set compile flags etc. for applications written in Nim (draft).

https://github.com/centurysys/nuttx-apps/blob/MAS1xx/master/config.nims

closed 2023-02-28 by centurysys