apache/nuttx #8538: Kconfig:249: warning: the 'modules' option is not supporte

pkarashchenko, 2023-02-15:
I'm using Ubuntu:focal as a host system and after recent sync with mainline I have next behavior on attempt to configure the build:
```
make menuconfig 
CP: arch/dummy/Kconfig to /home/pkarashchenko/eclipse-workspace/nuttx-upstream/nuttx/arch/dummy/dummy_kconfig
CP: boards/dummy/Kconfig to /home/pkarashchenko/eclipse-workspace/nuttx-upstream/nuttx/boards/dummy/dummy_kconfig
LN: platform/board to /home/pkarashchenko/eclipse-workspace/nuttx-upstream/apps/platform/dummy
LN: include/arch to arch/arm/include
LN: include/arch/board to /home/pkarashchenko/eclipse-workspace/nuttx-upstream/nuttx/boards/arm/samv7/same70-qmtech/include
LN: drivers/platform to /home/pkarashchenko/eclipse-workspace/nuttx-upstream/nuttx/drivers/dummy
LN: include/arch/chip to /home/pkarashchenko/eclipse-workspace/nuttx-upstream/nuttx/arch/arm/include/samv7
LN: arch/arm/src/chip to /home/pkarashchenko/eclipse-workspace/nuttx-upstream/nuttx/arch/arm/src/samv7
LN: arch/arm/src/board to /home/pkarashchenko/eclipse-workspace/nuttx-upstream/nuttx/boards/arm/samv7/same70-qmtech/../common
LN: arch/arm/src/board/board to /home/pkarashchenko/eclipse-workspace/nuttx-upstream/nuttx/boards/arm/samv7/same70-qmtech/src
Kconfig:249: warning: the 'modules' option is not supported. Let me know if this is a problem for you, as it wouldn't be that hard to implement. Note that modules are supported -- Kconfiglib just assumes the symbol name MODULES, like older versions of the C implementation did when 'option modules' wasn't used.
warning: ESP32C6_DEVKIT_RUN_IRAM (defined at boards/risc-v/esp32c6/esp32c6-devkit/Kconfig:8) has 'help' but empty help text
Loaded configuration '.config'
No changes to save (for '.config')
```

xiaoxiang781216, 2023-02-15:
it's a known kconfiglib issue: https://github.com/ulfalizer/Kconfiglib/pull/119, but kconfig-frontend doesn't report this warning.

pkarashchenko, 2023-02-15:
I use `kconfig-frontend` installed with `apt` and not one from `tools` repo.

anchao, 2023-02-16:
Fixed by PR https://github.com/apache/nuttx/pull/8504 https://github.com/apache/nuttx/pull/8504/commits/d9789aea31fee04523d97b07d20dabfc9407f98e

closed 2023-02-17 by GUIDINGLI