apache/nuttx #8551: STM32F4 Ethernet descriptors alignment

fjpanag, 2023-02-16:
I am having a look on the Ethernet driver of the STM32F4's.

I am looking at the [reference manual](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&cad=rja&uact=8&ved=2ahUKEwiW3pLpiJv9AhWwQvEDHYQFBgEQFnoECBQQAQ&url=https%3A%2F%2Fwww.st.com%2Fresource%2Fen%2Freference_manual%2Fdm00031020-stm32f405-415-stm32f407-417-stm32f427-437-and-stm32f429-439-advanced-arm-based-32-bit-mcus-stmicroelectronics.pdf&usg=AOvVaw2x8tbTRz8d9PfqXBk3qZ74), and I thing there are some issues with the alignment of the DMA descriptors.

Firstly, in page 1226, it states that the DMA descriptors need to be word-aligned (32-bits):

> The Transmit descriptor list address register points to the start of the transmit descriptor list.
> The descriptor lists reside in the STM32F4xx's physical memory space and **must be word-
> aligned**. The DMA internally converts it to bus-width-aligned address by taking the
> corresponding LSB to low. Writing to the ETH_DMATDLAR register is permitted only when
> transmission has stopped. Once transmission has stopped, the ETH_DMATDLAR register
> can be written before the transmission Start command is given.

*(Emphasis mine.)*

The same note applies for both Tx & Rx.

I see that the descriptors are defined [here](https://github.com/apache/nuttx/blob/master/arch/arm/src/stm32/stm32_eth.c#L640), without any care for their alignment.

Isn't this wrong?  
What is the preferred way in NuttX to align such structures, is `__attribute__ ((aligned (4)))` OK?  
(Thinking of portability between compilers).

---

The same seems to apply to the buffers that the descriptors point to.

They are declared [here](https://github.com/apache/nuttx/blob/master/arch/arm/src/stm32/stm32_eth.c#L645), again without any care for alignment.

Indeed, the manual says that they can be byte-accessed (page 1165):

> **33.6.3 Host data buffer alignment**
> The transmit and receive data buffers do not have any restrictions on start address
> alignment. In our system with 32-bit memory, the start address for the buffers can be aligned
> to any of the four bytes. However, the DMA always initiates transfers with address aligned to
> the bus width with dummy data for the byte lanes not required. This typically happens during
> the transfer of the beginning or end of an Ethernet frame.

However, a very interesting note follows in page 1166:

> Note: Even when the start address of a receive buffer is not aligned to the system databus width
> the system should allocate a receive buffer of a size aligned to the system bus width. For
> example, if the system allocates a 1024 byte (1 KB) receive buffer starting from address
> 0x1000, the software can program the buffer start address in the receive descriptor to have
> a 0x1002 offset. The receive DMA writes the frame to this buffer with dummy data in the first
> two locations (0x1000 and 0x1001). The actual frame is written from location 0x1002. Thus,
> the actual useful space in this buffer is 1022 bytes, even though the buffer size is
> programmed as 1024 bytes, due to the start address offset.

This seems only as a good cause for issues.

---

It is true that the Ethernet peripheral on the STM32F4's is working as is.  
But for me, it is very unstable. I experience lot's of unexpected behaviours, communications problems, and occasionally complete failure of any Ethernet activity.

They are somewhat "rare" events, but they are there. Rare enough to be extremely difficult to troubleshoot, but frequent enough to make the device very problematic on its actual purpose.

Alignment issues would make sense for such problems I guess...

fjpanag, 2023-02-16:
@davids5 IIRC you are making heavy use of the Ethernet on the STM32F4's, and you are experienced on this peripheral.

Do you have any suggestion on this issue?  
Shall I provide a PR with aligned buffers, or I am missing something here?

davids5, 2023-02-17:
@fjpanag  No not on the F4 per say

We are using F7 and H7 with dcache so the alignment is done there,:

See https://github.com/apache/nuttx/blob/master/arch/arm/src/stm32f7/stm32_ethernet.c#L657-L660

See https://github.com/apache/nuttx/blob/master/arch/arm/src/stm32h7/stm32_ethernet.c#L660-L661 

That should answer the  question you posed
>What is the preferred way in NuttX to align such structures, is __attribute__ ((aligned (4))) OK?
(Thinking of portability between compilers).