apache/nuttx #8557: Kcongilib can not build run menuconfig

davids5, 2023-02-17:
@xiaoxiang781216 After  https://github.com/apache/nuttx/pull/8494

tools/configure.sh nucleo-h743zi:nsh

make menuconfig

```
Loaded configuration '.config'
Traceback (most recent call last):
  File "/usr/lib/python3.8/curses/__init__.py", line 89, in wrapper
    cbreak()
_curses.error: cbreak() returned ERR

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/david_s5/.local/bin/menuconfig", line 8, in <module>
    sys.exit(_main())
  File "/home/david_s5/.local/lib/python3.8/site-packages/menuconfig.py", line 663, in _main
    menuconfig(standard_kconfig(__doc__))
  File "/home/david_s5/.local/lib/python3.8/site-packages/menuconfig.py", line 732, in menuconfig
    print(curses.wrapper(_menuconfig))
  File "/usr/lib/python3.8/curses/__init__.py", line 111, in wrapper
    nocbreak()
_curses.error: nocbreak() returned ERR
```

Do we need to check packages and emit instruction to unblock users?

tmedicci, 2023-02-17:
Same here!

The problem was introduced by https://github.com/apache/nuttx/commit/fff0e588600d72443b543e1920acd05f91aeb1e4

Piping to `grep` throws the issue. Temporarily, revert the commit, please.

xiaoxiang781216, 2023-02-18:
@anchao could you take a look?

anchao, 2023-02-20:
Please review PR: https://github.com/apache/nuttx/pull/8586

closed 2023-02-20 by xiaoxiang781216