apache/nuttx #8578: initial configuration problem after moving local files

cederom, 2023-02-18:
Hello world :-)

I found a problem with initial configuration after some git remote renames and top `nuttxworkspace.git` directory. No clue why exactly :-)

```
% git checkout master
Switched to branch 'master'
Your branch is up to date with 'origin/master'.
octagon% git pull --all
Fetching origin
Fetching cederom
Already up to date.

% ./tools/configure.sh -E -B sim:nsh
gmake: Entering directory '/XXX/nuttxworkspace.git/nuttx'
tools/Unix.mk:33: /XXX/nuttxworkspace.git/nuttx/Make.defs: No such file or directory
gmake: *** No rule to make target '/XXX/nuttxworkspace.git/nuttx/Make.defs'.  Stop.
gmake: Leaving directory '/XXX/nuttxworkspace.git/nuttx'

% gmake distclean
tools/Unix.mk:33: /XXX/nuttxworkspace.git/nuttx/Make.defs: No such file or directory
gmake: *** No rule to make target '/XXX/nuttxworkspace.git/nuttx/Make.defs'.  Stop.

% gmake clean
tools/Unix.mk:33: /XXX/nuttxworkspace.git/nuttx/Make.defs: No such file or directory
gmake: *** No rule to make target '/XXX/nuttxworkspace.git/nuttx/Make.defs'.  Stop.

% gmake
tools/Unix.mk:33: /XXX/nuttxworkspace.git/nuttx/Make.defs: No such file or directory
gmake: *** No rule to make target '/XXX/nuttxworkspace.git/nuttx/Make.defs'.  Stop.
```

* Okay, so the `Make.defs` seems missing:

```
% ls -al Make.defs
lrwxr-xr-x  1 cederom  cederom  115 18 lut 23:18 Make.defs -> /XXX/nuttx.git/nuttx/tools/../boards/sim/sim/sim/scripts/Make.defs
```

* Yup, thats the old directory location that does not exist.
* So I have removed the `Make.defs` file.
* `tools/configure.sh` script still complains file is missing.

```
% ./tools/configure.sh -E -B sim:nsh
gmake: Entering directory '/XXX/nuttxworkspace.git/nuttx'
tools/Unix.mk:33: /XXX/nuttxworkspace.git/nuttx/Make.defs: No such file or directory
```

Silly question: How to recreate this file? Isn't that the job of the `tools/configure.sh` script? :-)

I am using `gmake` because on BSD `make` is the BSD make, while `gmake` is the GNU make that NuttX requires. Will try to fix that one day :-)

cederom, 2023-02-18:
* Okay so I had to simply link the default `Make.defs` to some board config and things work back again :-)
* Do we want to fix that situation shomehow in the `tools/configure.sh`?

```
% ln -s boards/sim/sim/sim/scripts/Make.defs Make.defs

% gmake
LN: include/arch to arch/sim/include
LN: drivers/platform to /XXX/nuttxworkspace.git/nuttx/drivers/dummy
LN: arch/sim/src/chip to /XXX/nuttxworkspace.git/nuttx/arch/sim/src/sim
LN: arch/sim/src/board to /XXX/nuttxworkspace.git/nuttx/boards/sim/sim/sim/src
(...)

octagon% gmake clean
```

xiaoxiang781216, 2023-02-21:
The link ensure the update to the original file doesn't need to run configure.sh again.

cederom, 2023-02-21:
Thanks @xiaoxiang781216 :-) 

Additional check if file is valid (exist?) would be nice to have, warn that configure needs to be re-run if invalid, maybe even auto-re-run it..? :-)

The problem is that re-running `tools/configure.sh` does not fix the situation.. for a newcomer this may be a bit confusing :-P

xiaoxiang781216, 2023-02-22:
Sure.