apache/nuttx #8579: Add BSD include paths (/usr/local/include).

cederom, 2023-02-18:
Hello world :-)

I have tried to build the `sim:nxwm` example, but got this on FreeBSD:

```
./sim/posix/sim_x11framebuffer.c:29:10: fatal error: 'X11/Xlib.h' file not found
#include <X11/Xlib.h>
         ^~~~~~~~~~~~
1 error generated.
```

This file exist but in the `/usr/local` include prefix: 

```
% ls -al /usr/local/include/X11/Xlib.h
-rw-r--r--  1 root  wheel  99811  1 gru 02:14 /usr/local/include/X11/Xlib.h
```

What would be the best way to add this path to global includes? :-)

xiaoxiang781216, 2023-02-19:
you can add the search to HOSTCFLAGS here:
https://github.com/apache/nuttx/blob/master/arch/sim/src/Makefile#L142-L156
But it's strange that /usr/local/include/ isn't on your default toolchain search path.

cederom, 2023-02-19:
Thank you @xiaoxiang781216 that should provide a quickfix :-)

Here on BSD we usually include `.include <bsd.prog.mk>` that holds all defaults.. but NuttX uses GNU make so this won't work.. unless I find elegant workaround to use both BSD and GNU make files :-P

cederom, 2023-02-19:
Its alive! :-)

```
% gmake
LD:  nuttx
ld: warning: arch_setjmp_x86_64.o: missing .note.GNU-stack section implies executable stack
ld: NOTE: This behaviour is deprecated and will be removed in a future version of the linker

% ./nuttx
nxwm_main: Initialize the NSH library
createTaskbar: Create CTaskbar instance
createTaskbar: Connect CTaskbar instance to the NX server
createTaskbar: Initialize CTaskbar instance
createStartWindow: Start the start window application
createNxTerm: Creating the NxTerm application
createNxTerm: Adding the NxTerm application to the start window
createHexCalculator: Creating the hex calculator application
createHexCalculator: Adding the hex calculator application to the start window
startWindowManager: Start the window manager
```

pkarashchenko, 2023-02-19:
@cederom Congrats! Do we need any patch to NuttX or that is purely host system tuning?

cederom, 2023-02-19:
Thanks @pkarashchenko :-)

* Its more about initial build configuration ( `tools/configure.sh`) that would tune the tools on BSD.
* I have already provided some initial patches to select BSD (with `-B` switch) but more needs to follow :-P
* This particular issue presented above relates only to `nxwm` example, that also uses custom includes/libs location for macOS, the same can be done for BSD, or we can simply add this location top level so it is visible to all applications and none of them needs to be updated separately.
* I will provide the patches when ready :-)
* We can keep this issue open until pr is provided, or close and I will report back here when pr is ready, as you please :-)
* I am finishing a bigger project right now, but also starting a new one where I want to use NuttX, so my full focus should go to NuttX + BSD soon, sorry it takes so long :-)