apache/nuttx #8591: MMIO access without going through HAL/drivers

szsam, 2023-02-20:
I have noticed that some code accesses memory-mapped I/O (MMIO) without going through Hardware Abstraction Layer (HAL) or drivers. This type of access may have several drawbacks, including poor portability and reusability.

I would like to open this issue to investigate why some MMIO access does not follow the conventional path through HAL or drivers. Are there specific scenarios where non-conventional MMIO access is preferred or a must (e.g. because, for example, SDK does not provide the necessary API)? Is it possible to fix the code by accessing MMIO conventionally (i.e. through HAL/drivers)?

Code snippets with MMIO access that does not go through HAL or drivers:
https://github.com/apache/nuttx/blob/f0825c240259a53753ee5ca38c7e2e4bf09a7d07/sched/init/nx_bringup.c#L267

xiaoxiang781216, 2023-02-20:
> I have noticed that some code accesses memory-mapped I/O (MMIO) without going through Hardware Abstraction Layer (HAL) or drivers. This type of access may have several drawbacks, including poor portability and reusability.
> 

MMIO is normally access through macros defined in arch specific header file:
https://github.com/apache/nuttx/blob/master/arch/arm/src/common/arm_internal.h#L130-L141

> 
> Code snippets with MMIO access that does not go through HAL or drivers:
> 
> https://github.com/apache/nuttx/blob/f0825c240259a53753ee5ca38c7e2e4bf09a7d07/sched/init/nx_bringup.c#L267

this is a normal memory region, not specific to MMIO.

acassis, 2023-02-20:
@szsam that is the NuttX's compromise! Everything needs to use the drivers and POSIX abstraction, this way an application that works on MCU A will also work on MCU B, C, D, ... and Z.