apache/nuttx #8604: Which implementation of NuttX select/poll/EPOLL is recommended in terms of performance and efficiency?

centurysys, 2023-02-21:
I'm implementing the **Nim language** for NuttX.

NuttX has select/poll/EPOLL available as IO wait mechanisms, which one is recommended in terms of performance and efficiency?
However, since EPOLL is implemented in Nim with a minimum preallocation of 1024 fd's (although I intend to use OPEN_MAX as the upper limit), and since EPOLL in NuttX seems to use poll() internally,  I felt that "select" would be fine if there is no benefit as EPOLL on Linux.

centurysys, 2023-02-21:
It does not have to be an exact measurement. It is sufficient to know which one is desirable for the design.

xiaoxiang781216, 2023-02-21:
After https://github.com/apache/nuttx/pull/7642 epoll is fast then poll/select.

centurysys, 2023-02-21:
@xiaoxiang781216 
Thank you for your prompt reply !
Now I will try to use EPOLL as much as possible.

closed 2023-02-21 by centurysys